arduino/Arduino #2656: Link original Arduino bootloaders with boards.txt

NicoHood, 2015-02-17:
uno.bootloader.file=arduino:optiboot/optiboot_atmega328.hex

does not work

```
uno.name=Arduino Uno HID-Bridge

#uno.vid.0=0x2341
#uno.pid.0=0x0043
#uno.vid.1=0x2341
#uno.pid.1=0x0001

uno.upload.tool=arduino:avrdude
uno.upload.protocol=arduino
uno.upload.maximum_size=32256
uno.upload.maximum_data_size=2048
uno.upload.speed=115200
uno.upload.use_1200bps_touch=true
uno.upload.wait_for_upload_port=true

uno.bootloader.tool=arduino:avrdude
uno.bootloader.low_fuses=0xFF
uno.bootloader.high_fuses=0xDE
uno.bootloader.extended_fuses=0x05
uno.bootloader.unlock_bits=0x3F
uno.bootloader.lock_bits=0x0F
uno.bootloader.file=arduino:optiboot/optiboot_atmega328.hex

uno.build.mcu=atmega328p
uno.build.f_cpu=16000000L
uno.build.board=AVR_UNO
uno.build.core=HID:hid

uno.build.variant=standard
```

Wackerbarth, 2015-09-28:
@cmaglie -- This would really help simplify things and cut down on multiple copies of these boot files all over the place.

per1234, 2017-04-05:
Just wanted to add the correct terminology so this issue can be found in searches more easily:

reference or referencing bootloader

This would be a really useful feature.