arduino/Arduino #2663: Upgrade from v 1.05 to v 6.0 issue.

daddyusa17, 2015-02-19:
Once I saw a new upgrade Arduino version from 1.05 to 6.0, I did immediately downloaded the new version. Guess what? My old important cool projects had errors after complied them. I was shocked and panic then I went back to old perfect 1.05 version, hopefully all of my sketches will work just like before and in the past. All were working perfectly. Now nothing is perfect. I lost everything! I am thinking about next move by buy Raspberry Pi. No more errors. 
Guys, same problem like mine when installed 6.0 from any old version?
Please help me to get my old sketches back to work and fun to play with even proud of my works.
How?

ffissore, 2015-02-19:
Your sketches are in your sketchbook, so you haven't lost anything. What's your sketchbook path? Are there error messages?

z3t0, 2015-02-25:
As mentioned above, upgrading your IDE should not change your existing sketches and they should be in the sketchbook folder. 

1. What Operating System are you on?
2. Where did you save your sketches? (Were they saved in the default sketchbook location?)
3. Check the recycle bin in case it got moved there.

Are you using the default sketchbook path?

daddyusa17, 2015-02-25:
***1. Windows 7 - 64bit
 
***2. I saved my sketches into a new folder of ARDUINO 2015 before upgraded to version 6.0.
    Then I put my old sketches back to default sketchbook location.
In the past, I learned my first mistake that I lost everything when upgrade version issue so I transferred all files into USB drive for backup in case and I always do.
 
***3. I still saved my old sketches though.
 
I upgraded version 6.0 for first time from version 1.05. There were so many errors that most sketches doesn't work. Of course, I was panic then I uninstalled new version and back to v1.05. More problems. I went back to version 6.0 for a few days then I uninstalled it and back to version 1.05. Voila. All sketches are working okay. 
 
Now I am afraid go back to version 6.0 so I stay with version 1.05 and I am happy so far so good.
 
 

 
Date: Tue, 24 Feb 2015 20:40:41 -0800

z3t0, 2015-02-25:
That's interesting hmm. Sorry I can't really help because everything should be working just fine, considering 1.6.0 is a stable release, can you give us a small example sketch that did not work?

daddyusa17, 2015-02-26:
Check it out below that's how I got the code.
 
https://raw.githubusercontent.com/twonk/RGBPongClock/master/Arduino/RGBPongClock/RGBPongClock.ino
 
I haven't installed a new Arduino W5100 internet shield yet due to no micro sd card.
I have no idea what it looks like on weather data on my project.
 
Videos of the clock in action:
https://vine.co/v/hwML6OJrBPwhttps://vine.co/v/hgKWh1KzEU0https://vine.co/v/hgKz5V0jrFn 
 
 
It is so cool and one day I shall have it. Check it out below.
 
https://123led.wordpress.com/about/
 
 
 
 
It works fine on Arduino IDE v 1.05 but too many errors on Arduino IDE v 1.6.0
See if you can make it work on Arduino IDE v 1.60
Compare it with Arduino IDE v 1.05
I use Mega 2560 R3.
 
 
 
I know RGB Pong Clock sketch is not perfect due to font sizes that hard to read and other issues on pong, DS1307 issue like turn off the power and turn back it on, I have to rerset time over again and again even I know the wiring diagram perfectly since I tried on binary clock sketch, no problem. Overall, I love it so much though even I work very hard to make fonts readable. So far, no luck. I do not know how to change 3x5 and 5x5 fonts in the library. Even no help from internet guys on line. Sigh. Very frustrate to see like that any code I download and paste it on sketch them to verify but errors especially library issues and other errors as well so I browsed to find a certain file but no luck. Sometime it works but most sketches from internet which has too many errors. I am not happy about it.
 
Even though I keep buying Arduino stuffs. O well.  LOL.
 
Thank you for your time trying to figure it out about Arduino version issue.
I appreciate it very much.
 
Terry.
 
 

 
Date: Wed, 25 Feb 2015 14:57:50 -0800

daddyusa17, 2015-02-26:
Oops. I forgot to mention that I use one ADAFRUIT 16x32 RGB panel on RGB Pong Clock sketch. Very beautiful look.
I got two 64x32 Dot Matrix Red Green LED Display Panel in mail today.   :-)

 
Date: Tue, 24 Feb 2015 20:40:41 -0800

z3t0, 2015-02-26:
Thank you very much for your highly informative replies, hopefully we will soon be able to find a solution.

If you wouldn't mind could you compress the external libraries needed for this.

I was trying to get the libraries myself but soon realized that there were far too many to collect.

Please share them here so it is easier to test on the latest IDE

Thanks!

ffissore, 2015-02-27:
@daddyusa17 can you paste the error message you get? The only clear thing so far is that "it's not working". If you can paste the error message we can understand why and what causes the problem

bperrybap, 2015-03-11:
All the libraries other than "blinky" are easy to find.
Where is the blinky  library? (Assuming it is a library)
Where are the font header files coming from?

Anyway, when I comment out the includes for blinky.h and the 2 fonts it makes it far enough
to see what is likely causing the problem - which is what my first guess would have been.

It appears the issue is because the newer IDEs use newer versions of the
 AVR compiler and AVR libC and  and the newer compilers are more strict on progmem references. 
Assuming this is the case, the code will need to be corrected to work the newer
versions of the compiler.

Unfortunately incorrect progmem references worked with the older compiler but don't work
with the new compiler. Also some of the prog types have been eliminated in the newer
version of AVR libC.

What made things a bit worse for this, is that for several years, up until recently,
the examples on the Arduino progmem web page were using incorrect progmem declarations. 
While they worked with the older compiler they fail with the newer versions.
Because of this, there is likely to be lots of code that incorrectly used
the AVR progmem/PROGMEM compiler hack which is going  to break on the newer compilers.

The good news is that onces the declarations and progmem types are corrected,
the code will compile on the newer and older compilers.

--- bill

daddyusa17, 2015-03-11:
Hi Bill.
 
Guess what?
 
It was a bad wiring that caused bad fonts so I decided to take off all wirings and started all over again and inserted one by one very carefully through ADAFRUIT forum. Voila. The fonts were perfect to read. I am happy. Only one problem, I had a hard time to make Ethernet W5100 worked but no weather data. I don't know how to set up a static ip for RGB pong clock project. Even I read the article about how to set up a static ip but too hard for me to understand even AT&T u-verse offered me to buy static ip from them but I read the article stated don't buy. So I give up. Everything is working wonderfully except no weather data. Strangely, I have a binary clock project with DS1307 kit from ADAFRUIT. It works perfectly even the power off and hooked up again and the time is still reading properly. On RGB pong clock project, I opened the serial communication monitor at 9600 and it failed to read DS1307 even a bad circuitry from an example out of ARDUINO IDE code told me so. Weird? Perhaps I need a different DS1307 kit other than ADAFRUIT or different code and library as well to read time properly. What's more if I use DS1307 for RGB Pong Clock project. The time is running very well. Strange, isn't it?
Right now no DS1307, it is running alright with one or two minutes error daily. Not bad.
Overall, I like to see weather data. I just put them away. I need my mind rest. LOL.
 
//Setup networking
//Had trouble with DHCP, so sticking with static
byte mac[] = { 
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
IPAddress ip( 192,168,0,210 );
IPAddress _dns( 192,168,0,2 );
IPAddress gateway( 192,168,0,1 );
IPAddress subnet( 255,255,255,0 );
 
//Weather Stuff
IPAddress server(192,168,0,73);
 
 
That code is from RGB Pong Clock project.
 
One day, I might success to make it working to read weather data. *crosses fingers*
 
I need to learn how to set up a static ip article.
 
http://portforward.com/networking/static-win7.htm
 
Thank you so much about old ARDUINO version and new version information. Even they asked me to upgrade from old to newer version. Of course, I stick to old version. No rush.
I wait till it comes up with better version that can read old compiles and old projects as well.
 
How did you find a math library? No luck for me but good thing that old version of ARDUINO IDE works. It would be very nice to add url address of each library on every code in any ARDUINO IDE project and all I have to do is click on it and get a library code easily. I know not all has that. 
 
Yeah, too many examples errors on new version. Even old version I use right now.
Sometimes I corrected it to make it works properly but most codes are terrible with errors.
 
Same thing with W5100 examples from ARDUINO IDE, very difficult because almost all examples are not working with old version. Right now it attach to Mega 2560 with RGB Pong Clock. I do not want to remove it so I plan to order another W5100 shield then I will try again with examples and check on wwwutube.com as well. I might learn something new about that.
 
 
 
Sorry, I talked a little too much.   :)
 
Terry.
 
 
 
 
 
 
 
 
 
 
 

 Yes, you are correct that blinky and fonts libaries are included in code of RGB Pong Clock.
 
Date: Tue, 10 Mar 2015 23:05:26 -0700

ffissore, 2015-05-21:
We won't provide the old version of the toolchain, so sketches and libraries can only 1) update their code 2) stick with 1.0.x which will remain available for download forever
Having @daddyusa17 found his way out, I'm closing this

closed 2015-05-21 by ffissore