arduino/Arduino #2665: Serial.println("'\n'); causes stack overflow

tjnomad, 2015-02-19:
this statement: Serial.println("'\n'); 

causes a stack overflow.  Obviously the missing " is not caught before the line ends!

Your welcome ...TJ

TJ Schmidt: entrepreneur, scientist, creator of wonderful stuff!

ffissore, 2015-02-19:
It looks like a preprocessor issue. Please give a spin to the version of the IDE linked at #2636 (near the bottom of the conversation)

Chris--A, 2015-05-21:
This has been resolved, the code makes it to the compiler and produces an error as expected.

IDE 1.5.7
```error
sketch_may21a.ino: In function 'void setup()':
sketch_may21a.ino:6:3: error: 'Serial' was not declared in this scope
sketch_may21a.ino:7:1: error: expected primary-expression before '}' token
sketch_may21a.ino:7:1: error: expected ';' before '}' token
```

IDE 1.6.5 (source build)

```gcc
sketch_may21a.ino:3:3: error: missing terminating " character
sketch_may21a.ino: In function 'void setup()':
sketch_may21a.ino:4:1: error: expected primary-expression before '}' token
sketch_may21a.ino:4:1: error: expected ';' before '}' token
missing terminating " character
```

Chris--A, 2015-05-26:
Does this still cause problems on operating systems other than windows?

ffissore, 2015-09-11:
Fixed by #3779

closed 2015-09-11 by ffissore

m-ri, 2016-09-11:
Ciao,

Ho  sentito che stavi cercando  quella roba da un lungo  periodo di tempo,  così l’ho finalmente trovata per te, maggiori informazioni qui  <http://pino.buenaparkdentalspa.com/vwgn>

Grazie per la tua considerazione,marco ribero