arduino/Arduino #2669: Arduino DUE - slow communication rate using Wire1

HanaJin, 2015-02-19:
I am using the Adafruit Motor Shield V2 library to control stepper motors. Their source code uses I2C communication under the hood to control the shield. The library uses Wire on Uno and Wire1 on Due, but both have the same default I2C bus speed. An important fact is that this is THE only difference between Uno and Due as far as the 3P library source code is concerned. 

However, the same command in the library is taking a lot more (hundreds or thousands times) time on Due than Uno. 

I have originally posted this question to Adafruit and am recommended by Adafruit to ask here. The original post is here 
https://forums.adafruit.com/viewtopic.php?f=31&t=68292&p=346159#p346159

Is there any know issue for Wire1 on Due? Thanks for any help.

sandeepmistry, 2015-10-08:
@HanaJin did you make any progress on this?

Unfortunately, I don't have access to the shield, but tried a Wire <-> Wire1 connection, connecting SCL0 to SCL1, SDA0 to SDA1, on my Due with the following sketch based on yours:

```arduino
#include <Wire.h>
#include <Adafruit_MotorShield.h>

Adafruit_MotorShield AFMS = Adafruit_MotorShield(); 
Adafruit_StepperMotor *myMotor = AFMS.getStepper(100, 2);

unsigned long time1;
unsigned long time2;

void setup() {
  Serial.begin(9600);

  Wire.begin(0x60);
  Wire.onReceive(receiveEvent);
  Wire.onRequest(requestEvent);

  Serial.println("AFMS begin");
  AFMS.begin();

  Serial.println("set speed");
  myMotor->setSpeed(200); 
}

void loop() {
  Serial.println("Move two steps MICROSTEP");
  
  time1 = millis();
  myMotor->step(2, FORWARD, SINGLE); 
  time2 = millis();
  
  Serial.print("Time used: ");
  Serial.println(time2 - time1);
}

void receiveEvent(int howMany) {
  while(Wire.available()) {
    char c = Wire.read();
  }
}

void requestEvent() {
  Wire.write(0x00);
}
```

It's reporting a time used of 12 or 13ms, so not really seeing any timing issues with Wire1 on the Due now.

@ladyada would someone from Adafruit have time to try out @HanaJin's sketch with the shield and a Due:
```arduino
#include <Wire.h>
#include <Adafruit_MotorShield.h>
#include "utility/Adafruit_PWMServoDriver.h"

Adafruit_MotorShield AFMS = Adafruit_MotorShield(); 
Adafruit_StepperMotor *myMotor = AFMS.getStepper(100, 2);

unsigned long time1;
unsigned long time2;

void setup() {
  Serial.begin(9600);         

  AFMS.begin();
  myMotor->setSpeed(200); 
}

void loop() {
  Serial.println("Move two steps MICROSTEP");
  time1 = millis();
  myMotor->step(2, FORWARD, SINGLE); 
  time2 = millis();
  Serial.print("Time used: ");
  Serial.println(time2-time1);
}
```

ladyada, 2015-10-08:
ok dont have this handy but off top of heade - its probably not i2c but rather the delayMicroseconds - is that known good on the Due? does it 'round up'?

PaulStoffregen, 2015-10-08:
delayMicroseconds was fixed quite some time ago.  Originally it was way off.  I remember, because it was a lot of trouble for people using OneWire, which I maintain.  I believe it was fixed by borrowing Teensy 3.0's code.  :)

PaulStoffregen, 2015-10-08:
Perhaps the slowness is in Serial?  It got transmit buffering only very recently.  I recommend testing this with the latest hourly build.  (which really should be common sense to do before filing *any* bug report with *any* open source project...)

sandeepmistry, 2015-10-09:
@ladyada @PaulStoffregen thanks for the info!

I was testing with the latest SAM core from Github, I think this issue can be closed, but it would be great to confirm with the real shield.

From @HanaJin's [post on the Adafruit forum](https://forums.adafruit.com/viewtopic.php?f=31&t=68292&p=346159#p345915), they were seeing a delay of 7 - 8ms on the Uno and ~2500 - 3300ms on the Due. This was back in Nov 2013.

HanaJin, 2015-10-09:
@sandeepmistry, can you share the delay number seen with the lastest SAM core? 
I tried 1.6.1 release which gave a number of ~1600ms on Due. On my old 1.6.6 nightly build (Sep 10), the number is ~860-1000ms. I also downloaded the lastest nightly build to try out, but was unable to get the same sketch compiled with multiple definitions error.

sandeepmistry, 2015-10-09:
@HanaJin thanks for the update. Unfortunately I don't have a motor shield to test the Due with, however when I run the first sketch from https://github.com/arduino/Arduino/issues/2669#issuecomment-146575363 (with the Wire1 connected to Wire port on the same Due) I get around 12 - 13 ms.

I was using version 1.0.1 of the Adafruit Motor Shield V2 library installed from the Library Manager, with the Due connected to the Programming Port on my Mac. With a 1.6.6 hourly build from today with the 1.6.4 of the Arduino SAM Boards package, also get 13ms for the same sketch.

closed 2017-02-08 by agdl