arduino/Arduino #2670: LiquidCrystal - "Hello World!" wires to pins 12 and 11 are reversed

sigh-must-sign-up, 2015-02-20:
http://arduino.cc/en/Tutorial/LiquidCrystal

This wiring image is wrong.
http://arduino.cc/en/uploads/Tutorial/LCD_bb.png

The blink code says to
LCD RS pin to digital pin 12
LCD Enable pin to digital pin 11

The image incorrectly wires 12 to R/W and 11 to RS.
See HD44780 spec, RS = 4, R/W = 5
Also, just try wiring like the image, and it won't work.
Switch pins 12 and 11, and it works okay.

VaclavSal, 2015-02-20:
Just comments.

While I share your concern for correct info  I cannot help to wonder WHY is a machine (?) generated wiring diagram so important. Apparently INSTRUCTIONS what to connect where and  / or schematic no longer work?
Does the average experimenter really follows the wiring diagram as presented instead of checking the schematic? Just asking. 

And  what is wrong correcting the "error" in software? 
But you are right, the error is in hardware so it should be corrected there.

BTW the R/W means "read / write" not "enable". 
But nobody uses LCD memory to read from anyway so it does not matter.

agdl, 2015-02-20:
Updated thank you

closed 2015-02-20 by agdl