arduino/Arduino #2672: Platform.h removed since Arduino IDE v1.5.8: Why?

pbony, 2015-02-21:
Hello,

I found 3 sketches online and all generate the same compilation error on arduino IDE v1.6.0:

C:\Program Files (x86)\Electronic\Arduino\libraries\EasyVR\EasyVRBridge.cpp:14:24: fatal error: Platform.h: No such file or directory
   #include "Platform.h"

To fix this issue, I had to download Arduino IDE v1.5.7 and copy/paste Platform.h into
C:\Program Files (x86)\Electronic\Arduino\hardware\arduino\avr\cores\arduino

So why Platform.h has been removed?
Would it be possible to include it again please?

Thanks for your inputs and feedback.
Best Regards

cmaglie, 2015-02-23:
The file has been removed since it merely includes other files and is not part of the documented API, here the relevant changeset:

https://github.com/arduino/Arduino/commit/2e3e539b718d3b809b1f374602b724124d8e97b5

To solve your issue you may try remove `#include <Platform.h>` from your sketch.

ppescher, 2015-02-23:
Please see comments on TIGAL/EasyVR-Arduino#1

cmaglie, 2015-02-23:
@pbony 

it seems that @ppescher has already fixed this in the latest development version of the library:

https://github.com/TIGAL/EasyVR-Arduino/archive/version-1.6.zip

please try it out and report back your findings here: https://github.com/TIGAL/EasyVR-Arduino/pull/1

closed 2015-02-23 by cmaglie