arduino/Arduino #2673: char array declaration problem in IDE v1.6.0

JPinRT, 2015-02-21:
I found a problem with declaring a char array in setup() in the latest release of IDE v1.6.0.
If I declare char tmpStr[18]=""; in setup(), tmpStr doesn't work when used within setup().  An example of not working would be strcpy(tmpStr, "Hello World");.  If I use Serial.println(tmpStr); before or after the strcpy function, the display is a series of characters similar to some type of comma.  However if I declare char tmpStr[ ="12345678901234567"; in setup(), tmpStr can be used fine.

If I declare tmpStr[18]=""; globally ( before setup() ), then tmpStr works fine.

The problem doesn't seem to exist in the previouse IDE v1.0.6.

JP

matthijskooijman, 2015-02-21:
Could you post a full, but minimal, sketch that shows the problem?

I wonder if your program might be triggering undefined behaviour that just accidentally works in previous gcc versions?

JPinRT, 2015-02-22:
I had replaced v1.6.0 back to v1.0.6 and continued with programming my little test project.  I did save the problem program though.
So I made a simple program to try to demonstrate the problem...

#include <RHReliableDatagram.h>  // include radiohead library
#include <RH_NRF24.h>  // more radiohead
#include <SPI.h>
#include <dht.h>
#include <U8glib.h>  // graphics library for OLED

// char tmpStr[18] = "";  // declaring here works in v1.6.0

void setup()	{
  char tmpStr[18] = "";  // declaring here this way does not work in v1.6.0
  //char tmpStr[18] = "123456789012345678";  // declaring here this way works in v1.6.0
  int counter = 1;

  Serial.begin(9600);
  Serial.println(tmpStr);

  do  {
    sprintf(tmpStr, "counter = %d", counter++);
    Serial.println(tmpStr);
    delay(2000);
  }  while  (  counter <= 50  );
}

void loop()  {
}

...re-installed v1.6.0, and the simple program worked fine.  I then retested the problem program, and it still has the problem.  I then tested the program I continued editing in v1.0.6 (removing my debug Serial.println's), and it works fine.
Very interesting.  I may have changed something as I was continue editing (trying to simplify some code), but the way I declared 'tmpStr' and used it had not changed.
I don't know, can't explain it, but it is working, so I guess this can be closed.

closed 2015-02-23 by ffissore