arduino/Arduino #2676: attachInterrupt page should document & recommend using digitalPinToInterrupt()

PaulStoffregen, 2015-02-22:
This page about attachInterrupt:

http://arduino.cc/en/Reference/AttachInterrupt

should document the new digitalPinToInterrupt() function and show an example, such as:

    attachInterrupt(digitalPinToInterrupt(pin), ISR, mode);

agdl, 2015-09-03:
@PaulStoffregen take a look to that page now please

agdl, 2015-09-08:
@PaulStoffregen I'm closing this. If you have further requests please reopen it.

PaulStoffregen, 2015-09-08:
Please edit this further.  I will supply suggested text in a few moments....

PaulStoffregen, 2015-09-08:
Edit 1: please move all but the first 2 sentences of Description to a new section titled "Interrupt Numbers" at the very end of the page.  I recommend this text:

**Interrupt Numbers**

Normally you should use digitalPinToInterrupt(pin), rather than place interrupt an number directly into your sketch.  The specific pins with interrupts, and their mapping to interrupt number varies on each type of board.  Direct use of interrupt numbers may seem simple, but it can cause compatibility trouble when your sketch is run on a different board.

However, older sketches often have direct interrupt numbers.  Often number 0 (for digital pin 2) or number 1 (for digital pin 3) were used.  The table below shows the available interrupt pins on various boards. 

        Board                                   int.0   int.1   int.2   int.3   int.4   int.5
        Uno, Ethernet                           2       3
        Mega2560                                2       3       21      20      19      18
        32u4 based (e.g Leonardo, Micro)        3       2       0       1       7
        Due, Zero                                               (see below)

The Arduino Due board has powerful interrupt capabilities that allows you to attach an interrupt function on all available pins. You can directly specify the pin number in attachInterrupt(). 

The Arduino Zero board allows you to attach an interrupt function on all available pins except for pin 4. You can directly specify the pin number in attachInterrupt().

PaulStoffregen, 2015-09-08:
Edit 2: Where you removed the text, add this new section.  It's important to place this simpler, recommended info at the top of the page.  The old, no-longer-recommended way should be located at the absolute *end* of the page.

**Digital Pins With Interrupts**

The first parameter to attachInterrupt is an interrupt number.  Normally you should use digitalPinToInterrupt(pin) to translate the actual digital pin to the specific interrupt number.  For example, if you connect to pin 3, use digitalPinToInterrupt(3) as the first parameter to attachInterrupt.

        Board:                                 Digital Pins Usable For Interrupts
        Uno, Nano, Mini, other 328-based       2, 3
        Mega, Mega2560, MegaADK                2, 3, 18, 19, 20, 21
        Micro, Leonardo, other 32u4-based      0, 1, 2, 3, 7
        Zero                                   all digital pins, except 4
        Due                                    all digital pins

PaulStoffregen, 2015-09-08:
Edit 3: In the Syntax section, make the digitalPinToInterrupt(pin) first, and edit the others to say they are not recommended.

    attachInterrupt(digitalPinToInterrupt(pin), ISR, mode);     (recommended)
    attachInterrupt(interrupt, ISR, mode)                       (not recommended)
    attachInterrupt(pin, ISR, mode)                             (not recommended: Arduino Due, Zero only)

PaulStoffregen, 2015-09-08:
Edit 4: In the Example section, edit the code sample.

      int pin = 13;
      volatile int state = LOW;

      void setup() {
        pinMode(pin, OUTPUT);
        attachInterrupt(digitalPinToInterrupt(pin), blink, CHANGE);
      }

      void loop() {
        digitalWrite(pin, state);
      }

      void blink() {
        state = !state;
      }

PaulStoffregen, 2015-09-08:
Please make these 4 edits.  I reopened this issue and supplied exact text, to make this as easy as possible for you.

I hope you can see how the small edit which adds digitalPinToInterrupt(), but leaves the old, no-longer-recommended approach prominently featured at the top of the page and still used in the examples is bad form.  Yes, it's quicker and easier to make only tiny edits to these old documentation pages.  But the result is poor documentation which still prominently features the old, bad way of writing code.

As you edit more Arduino documentation, I hope you will begin to consider the writing style.  Simply adding small notes tends to result in pages that are unclear about best practices.  I know this is a lot to ask, but please try to think of the effect this sort of documentation has for real Arduino users.  Especially when there are recommended ways to write code, Arduino's documentation should always explain the recommended way first.  All of the examples should be updated to the best way.  Older, poor ways should not remain in any examples.  Not-recommended approaches should always appear at the end of the page, as footnotes only for completeness of documentation.

Yes, this is a lot more work, but it is very important to do a good quality job on Arduino's documentation.

agdl, 2015-09-10:
@PaulStoffregen sorry fo the delay. Modified as you suggested. Many many thanks for the effort. Let me know if it is good please!

closed 2015-09-22 by agdl