arduino/Arduino #2677: Lockups on OSX

realrolfje, 2015-02-22:
I upgraded from Arduino 1.0.5 to 1.6 by copying the Arduino.app over the old one in the Applications folder (none of the Arduino code was running at the time, of course). This was on a Mac OSX 10.9.5 installation. Starting the new 1.6 IDE, I ran into 3 issues:

1. The Arduino 1.6 IDE does not copy the settings from 1.0.5, so the nice crisp font was replaced by an anti-aliased font which does not like the anti-aliasing as used in a Mac for some reason. Much "softer".
2. When opening a Sketch by double-clicking it, Arduino IDE opened 2 Sketches, the one I double clicked, and some empty demo project. This looks like an old 1.0.5 bug.
3. (the most important one, and the subject of this bug): The IDE intermittently locks up, or "hangs", for more than 15 seconds. It seems to happen the most when saving preferences or closing a project and exiting the IDE.

The third issue is clearly the most annoying one, and the reason I downgraded to 1.0.6 (yes, that's still one version better than the 1.0.5 I had). If there is anything I can do to add info or help please let me know, I really want to use 1.6 and move forward together with the Arduino community.

Thanks for hours of fun and a great piece of hardware/software. Very cool stuff, hope I can help (or somebody can help me) find out what causes my lockup problem.

PaulStoffregen, 2015-02-22:
Try turning off bluetooth.  Does that make any difference?

realrolfje, 2015-02-22:
Hi Paul,  Bluetooth is never on in this paranoid machine :-) Anything else I can try?

PaulStoffregen, 2015-02-22:
Maybe File > Prefs to turn off checks for updates?

roadfun, 2015-03-09:
Since you mention a preference issue when you upgrade, you might try deleting the preference file (it will create a new one).

From the Finder pick the Go menu then "Go to folder..." and enter ~/Library/Preferences. Then move the cc.arduino.Arduino.plist file to the trash (you can always get it back if you decided this doesn't help). Then run the IDE again.

I use the IDE a lot on Yosemite and haven't seen the sort of error you mention. But two other causes for pauses like this on a Mac: 1) An attached disk which has gone to sleep, the saving and opening actions sometimes cause such a drive to wake up and the Finder waits for it and 2) an attached folder/drive which is no longer available but the Finder looks for.

realrolfje, 2015-03-28:
I've installed 1.6.2 and the lockups seem to have been resolved. (Huge thanks!)

The "opening of two sketches" problem is changed but for the worse: When the Arduino ide is not running, and I double-click a sketch, the IDE opens an empty sketch, but not the one I double clicked.

When the IDE is running, and I double-click a sketch, it does open correctly.

ffissore, 2015-04-30:
Fixed in hourly builds. See #2888

closed 2015-04-30 by ffissore