arduino/Arduino #2680: 1.6.0 IDE verbose compile output misdirected while compiling with multiple projects open at same time

plocher, 2015-02-23:
MacOS Java6 IDE 1.6.0

1. Set preferences to verbose compiler output
2. Open 2 sketches in different windows (pick one that has several libraries to compile so the verify stage takes a little bit of time...)
3. Click on Verify on one of them 
4. While the compile is happening and generating progress output (i.e., before it finishes), click on the other sketch's window

BUG:  watch the first sketch's compile output be redirected to the second sketch's window's output region.

ffissore, 2015-02-23:
That's an old issue, although never reported before. Thanks, we'll fix it asap

ffissore, 2015-06-29:
This will be hard to fix. IDE uses plain access to stdout/stderr, which is intercepted and duplicated on both original stdour/err and current editor console. This means that the IDE is unable to understand the difference between something happening on one window and something happening on another window.

Solution is to replace each stdour/err call to a call to the local editor console. They are everywhere, so it will take time

plocher, 2015-06-29:
IMO, consider this more of a request for enhancement than a bug - I'm
pretty sure the sweet spot for users is still a single sketch...

That said, it is not unusual to see multi-arduino projects in the wild -
and those developers (myself included) would value better support for this
workflow...

Unfortunately, there are many obstacles to overcome, the biggest being that
the board settings are global instead of sketch specific, which makes it is
a royal pain to build a two-board system that uses different boards or usb
connections - the tools menu almost starts to melt with all the activity it
sees :-)

Since this isn't that request, and until a larger project addresses these
fundamental multi-IDE/board issues, this bug could easily be fixed in the
traditional way - with a release note - so it becomes, if not a feature, at
least not a surprise.

Peace,
  -John

lmihalkovic, 2015-11-27:
```
there are many obstacles to overcome
```
@plocher  the points you mention are real, however IMHO none of it a *show-stopper*

```
 They are everywhere, so it will take time
```
IMHE(xperience) a couple hours at best. maybe even a couple more to think up some missing bits. grand total 1/2 day at best. but say I am completely wrong, and it takes 1 full day. this *spring cleaning* will also address #4221 

before someone asks, my codebase has diverged a lot (looks more like a 2.0 at this point) last week when I added multi editor types (a match for [Arduino Create](https://blog.arduino.cc/wp-content/uploads/2015/05/Blogpost_f8.png), see #4197), but if not obvious (a subjective topic by all means) it is IMHE easy (measured by amount of code) to make these changes

lmihalkovic, 2015-12-04:
[LMDONE]