arduino/Arduino #2682: Using "arduino-tiny" and arduino-ide 1.6.0 experimental impossible on Mac OS X Yosemite.

stahlfabrik, 2015-02-23:
My Java Version is up to date. I run Yosemite.

Arduino IDE does not start after being closed when I selected my Attiny84:

Here is the boards.txt I use like the README of "tiny-0150-0020.zip" suggests:

```
attiny84at8.name=ATtiny84 @ 8 MHz  (internal oscillator; BOD disabled)

attiny84at8.upload.tool=arduino:arduinoisp

attiny84at8.upload.maximum_size=8192
attiny84at8.upload.maximum_data_size=512

# Default clock (slowly rising power; long delay to clock; 8 MHz internal)
# Int. RC Osc. 8 MHz; Start-up time PWRDWN/RESET: 6 CK/14 CK + 64 ms; [CKSEL=0010 SUT=10]; default value
# Brown-out detection disabled; [BODLEVEL=111]
# Preserve EEPROM memory through the Chip Erase cycle; [EESAVE=0]

attiny84at8.bootloader.low_fuses=0xE2
attiny84at8.bootloader.high_fuses=0xD7
attiny84at8.bootloader.extended_fuses=0xFF

attiny84at8.bootloader.path=empty
attiny84at8.bootloader.file=empty84at8.hex
attiny84at8.bootloader.tool=arduino:avrdude

attiny84at8.build.mcu=attiny84
attiny84at8.build.f_cpu=8000000L
attiny84at8.build.core=tiny
```

It is the taken from proposed boards from the zip file.

Now when I start Arduino IDE I can see the Attiny84 under "Tools/Board". But when I click on it, I get a Java Exception: 

```
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at processing.app.BaseNoGui.onBoardOrPortChange(Unknown Source)
	at processing.app.Base.onBoardOrPortChange(Base.java:1075)
	at processing.app.Base$4.actionPerformed(Base.java:1164)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.JToggleButton$ToggleButtonModel.setPressed(JToggleButton.java:308)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at com.apple.laf.ScreenMenuItemCheckbox.itemStateChanged(ScreenMenuItemCheckbox.java:193)
	at java.awt.CheckboxMenuItem.processItemEvent(CheckboxMenuItem.java:389)
	at java.awt.CheckboxMenuItem.processEvent(CheckboxMenuItem.java:357)
	at java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:351)
	at java.awt.MenuComponent.dispatchEvent(MenuComponent.java:339)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:754)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:702)
	at java.awt.EventQueue$3.run(EventQueue.java:696)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:724)
	at java.awt.EventQueue$4.run(EventQueue.java:722)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:721)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Now in the bottom right of IDE it still says "Arduino Yun", so nothing got changed to Attiny84. So I close the IDE. When I restart it, it does not come back up (no Window is appearing) - but the program is in the dock and the menu item "Arduino" is appearing in Mac OS X but then nothing happens - no window. When I rename my boards.txt to something else, I can start Arduino IDE again successfully. So the problem must be directly connected to the boards.txt of arduino-tiny.

I have a Debian 32 bit in virtual box. Here, with 1.6.0 the same boards.txt works just fine it seems (I get two lines on the console though, that I do not understand yet:
```
macmini-debian:~/Downloads/arduino-1.6.0$ sh arduino
+Die Platine tiny:avr:attiny84at8 definiert keine "Build.Platine"-Einstellung. Sie wurde automatisch auf AVR_ATTINY84AT8 gesetzt.
Die Platine tiny:avr:attiny84at1 definiert keine "Build.Platine"-Einstellung. Sie wurde automatisch auf AVR_ATTINY84AT1 gesetzt.

```
My locale is German so THAT is not the problem of understanding here;-) Is it related to the Mac OS issue?

What can I do about the Mac issue? How can I get rid of the NullPointer Exception?

PaulStoffregen, 2015-02-23:
Perhaps this is a duplicate of https://github.com/arduino/Arduino/issues/2635 ?

stahlfabrik, 2015-02-24:
The difference is that I did not remove the third party package: it does not even work like the readme tells me how to use it. 

And on Linux there is not the exception. Only on Mac os.

stahlfabrik, 2015-02-24:
Oh I just wanted to try the nightly build - it requests to have java 6, so this is NOT the experimental version, right?
Is there a nightly build for the experimental version, that fixes #2635?

stahlfabrik, 2015-02-24:
I am happy to report that I build arduino-ide from github. It runs on Java 7 AND does not NPE on choosing arduino-tiny board! Have not tried anything more - but it starts:-)

ffissore, 2015-02-24:
It looks like we forgot to add the link on the downloads page: that why it received so few downloads! :) Link added. Thanks

stahlfabrik, 2015-02-24:
That is a reasonable explanation ;-)

Will check it out in the next two days so this issue might be closed then.

If someone else wants to verify earlier please go ahead!

stahlfabrik, 2015-02-26:
This issue is solved in nightly build

closed 2015-02-26 by stahlfabrik