arduino/Arduino #2683: The explanation for example of delayMicroseconds() is not accurate

mikecat, 2015-02-24:
When I executed the example on [the reference of delayMicroseconds()](http://www.arduino.cc/en/Reference/DelayMicroseconds)([WebArchive](http://web.archive.org/web/20150224013927/http://www.arduino.cc/en/Reference/DelayMicroseconds))

> ```
> int outPin = 8;                 // digital pin 8
>
> void setup()
> {
>   pinMode(outPin, OUTPUT);      // sets the digital pin as output
> }
>
> void loop()
> {
>   digitalWrite(outPin, HIGH);   // sets the pin on
>   delayMicroseconds(50);        // pauses for 50 microseconds      
>   digitalWrite(outPin, LOW);    // sets the pin off
>   delayMicroseconds(50);        // pauses for 50 microseconds      
> }
> ```
>
> configures pin number 8 to work as an output pin. It sends a train of pulses with 100 microseconds period. 

on my Arduino UNO R3 and Arduino 1.0.6, the signal observed on pin 8 was approximately 9207Hz.
It means that the period is approximately 108.6 microseconds, and I think this error is caused due to the execution time of instructions other than ```delayMicroseconds()``` (i.e. ```digitalWrite()``` and the loop).

From this test, we can say that the sample do not send a train of pulses with 100 microseconds period but sends a train of pulses with **approximately** 100 microseconds period.
I think this explanation on the reference is a little bit misleading.
The problem becomes bigger when I change the delay from 50 microseconds to 5 microeconds: the output was approximately 55566Hz instead of expected 100kHz.

mikaelpatel, 2015-02-24:
@mikecat 
Did you try removing the delays? What would you expect then?

mikecat, 2015-02-24:
@mikaelpatel
I think it is a tough question to answer, but the result of my test on my Arduino was approximately 115160Hz.

agdl, 2015-09-04:
@mikecat I wrote this:

*It sends a train of pulses of approximately 100 microseconds period. The approximation is due to execution of the other instructions in the code.*

Is it ok for you? If so close the issue please otherwise tell me what you want to add to this page please

agdl, 2015-09-10:
@mikecat I'm closing this. Reopen it if needed please

closed 2015-09-10 by agdl