arduino/Arduino #2684: Confusing / inaccurate examples in "Port Registers" documentation

nickgammon, 2015-02-24:
On this page:

http://www.arduino.cc/en/Reference/PortManipulation

There is repeated reference to PORTD however the explanation below switches to PORT**B** and PIN**B** in a confusing way (bolded):

> Examples
> 
> Referring to the pin map above, the PortD registers control Arduino digital pins 0 to 7.
> 
> You should note, however, that pins 0 & 1 are used for serial communications for programming and debugging the Arduino, so changing these pins should usually be avoided unless needed for serial input or output functions. Be aware that this can interfere with program download or debugging.
> 
> DDRD is the direction register for Port D (Arduino digital pins 0-7). The bits in this register control whether the pins in PORTD are configured as inputs or outputs so, for example:
> 
> DDRD = B11111110;  // sets Arduino pins 1 to 7 as outputs, pin 0 as input
> DDRD = DDRD | B11111100;  // this is safer as it sets pins 2 to 7 as outputs
>                     // without changing the value of pins 0 & 1, which are RX & TX 
> 
> //See the bitwise operators reference pages and The Bitmath Tutorial in the Playground
> 
> **PORTB** is the register for the state of the outputs. For example;
> 
> PORTD = B10101000; // sets digital pins 7,5,3 HIGH
> 
> You will only see 5 volts on these pins however if the pins have been set as outputs using the DDRD register or with pinMode().
> 
> **PINB** is the input register variable It will read all of the digital input pins at the same time. 
>

agdl, 2015-02-27:
@nickgammon should be ok now

nickgammon, 2015-02-27:
Look good, thanks.

closed 2015-02-27 by nickgammon