arduino/Arduino #2685: "new" button behaviour is different from "file -> new"

ffissore, 2015-02-24:
Former will close the current sketch and replace it with a blank one. Latter will start a new sketch in a new windows
Stick with the latter behaviour

damellis, 2015-02-26:
I believe this was an intentional design decision that we inherited from Processing. They may have changed their minds about it, or we might prefer a different behavior -- but this isn't really a bug. The button is within the window and the new sketch is as well; the menu (on Mac in particular) is outside the window and so is the new sketch.

mastrolinux, 2015-02-26:
I can confirm this is different from what processing does (just tested). Processing does open a new window if you press the new icon and if you click on file -> open (same behaviour). We are doing something different. We have different behaviours for the same "new" action. This is confusing the users. Will ask UX designer about it.

PR #2723 (2015-03-02): NEW button now behaves as clicking File -> New menu entry.
Fixes #2685
/cc @mastrolinux @damellis
commit b7b8a354 (2015-03-02): NEW button now behaves as clicking File -> New menu entry. Fixes #2685
closed 2015-03-02 by mastrolinux