arduino/Arduino #2686: Windows driver issue with Arduino Due R3-E board

peabo7, 2015-02-24:
I recently purchased a bunch of Arduino Due boards whose programming ports are not recognized by the Windows 7 Arduino driver that is installed with 1.5.8 or 1.6.0.

The issue is that the USB vendor ID changed from 2341 (Arduino SA) to 2a03 (unknown).  The visible difference in the boards is that they are labeled in the back as from Arduino.org instead of www.arduino.cc and the board model is R3-E instead of R3.  (Also they have silkscreened pin names on the headers).  I assume these are legitimate boards, since I got them from a major distributor (Mouser?  I would have to check with purchasing to be sure).

The native port is unchanged and works fine, but I need the programming port so I can reset the Arduino.

So, I suppose I could reflash the 16u2 chip with a version of software that has the 2341 vendor ID, but I would rather not do that.

Currently I am exploring the possibility that I can provide a new arduino.inf file to make Windows 7 recognize the new vendor ID, but I wonder if anyone else has run into this problem or knows the pedigree of the R3-E board.  (I have other information about the problem but I haven't organized my observations yet in a way that I can report accurately.)

Peter Olson

ffissore, 2015-02-24:
Hi Peter, please forward your message to support@arduino.cc. My colleagues will contact you right after

closed 2015-02-27 by ffissore

peabo7, 2015-03-02:
> On February 24, 2015 at 10:26 AM Federico Fissore <notifications@github.com>
> wrote:
> 
> Hi Peter, please forward your message to support@arduino.cc. My colleagues
> will contact you right after

The support desk pointed me to a download 1.5.8.1 which solved my problem with
the new boards.

Will this fix appear in 1.6.0.x soon?  I have a bug in my program which I need
to solve in order to work with 1.6.0, but I am eager to try its new features.
 I'm not quite sure what the problem is, but in 1.5.8 the compiler was
misbehaving by interpreting a pointer to a device register as a pointer into
RAM.  I hacked around it, but in 1.6.0 the new ld program diagnoses an error
that I am trying to reference data which is not in the RAM segment, which must
be related.

Peter Olson

ffissore, 2015-03-11:
Version 1.6.1 should solve your issue