arduino/Arduino #2687: internal pullups on I2C

cano64, 2015-02-24:
This is a serious bug, that may cause destruction of a hardware attached to I2C and has been known for many years, yet it has not been fixed.

Arduino/hardware/arduino/avr/libraries/Wire/utility/twi.c
those two lines needs to be commented out.
digitalWrite(SDA, 1);
digitalWrite(SCL, 1);

If one has a I2C device that is powered by 3.3V and is not 5V tolerant, using the current Wire library will cause 5V to be applied through the internal pullup resistors of the Atmel part to the I2C device, possibly destroying the device.

setting internal pullup resistors on I2C has absolutely no benefits whatsoever. Most I2C devices need pullup resistors of 4.7k-10k ohms, internal pullups are 20k ohms or higher and are no good for this purpose.
If one wishes to use internal pullups for some reason, they can set them manually in their sketch.

PaulStoffregen, 2015-02-24:
Can you give even 1 specific example of a part that is actually destroyed?  Just 1 part number?

I would really like to test this myself, so if you do name a part number, I will buy one and personally test this.

So far, pretty much all the not-5V-tolerant parts I've encountered are not 5V tolerant because they have a ESD protection diode between their pin and the 3.3V power line.  Applying more than 3.8 volts causes conduction through that diode.  But the tiny current which flows as a result is typically well within the safe range for that internal ESD diode to conduct.

But perhaps there are some I2C chips with some real susceptibility?  Please, if you know of any such part that is actually damaged, please mention the part number.  I will physically test, so please make sure you know it's a part that is actually damaged, not merely a guess that it might be damaged in theory (under a theory that going even slightly over 3.3V with even the smallest current results in problems).

cmaglie, 2015-08-27:
Closing as duplicate of #506

closed 2015-08-27 by cmaglie