arduino/Arduino #2689: Due problem with pulseIn()....

Bartje64, 2015-02-25:
I originally had written some test code for the Uno that utilizes pulseIn(). Now I want to run this on a Due.

It works fine on the Uno, but refuses to on the Due. 
Thinking it was my code, I chased everything to find nothing. Tried another Due, same thing.
Simplified the code...basically only has the pulseIn() statement and a serial print of the result.
Nothing....

Lastly I tried it on a Mega, and it works just as it did on the Uno.

This leads me to believe that the problem lies somewhere within with the Due boards.

Anyone else have the same problem?
Is this a known issue?
Any fixes?

z3t0, 2015-02-25:
Please make sure you are using the latest IDE (1.6.0) as DUE is a relatively new board.
http://arduino.cc/en/Main/Software

cmaglie, 2015-02-25:
@Bartje64 
Please post the FULL sketch and a detailed description of the problem to help us understand what's going on. 

Thanks.

Bartje64, 2015-02-25:
I am using the latest IDE  as suggested....
The full sketch is 450 lines long, this test sketch does the same thing.

I have an IR sensor on my power meter. The pulse (~25ms in duration), pulls the input low.
I installed  a 1uf cap from gnd to input pin in order to stabilize and clean up the signal.
With the cap installed the Uno and Mega report  a time of ~40ms. This concurs with the readings I took with my scope.
The leading edge of the signal (gets pulled low), is straight down to 0V, trailing edge slopes up (due to caps recharging).
This same set-up does not work on the Due.

//**********************code begin******************************//
int IR_Sensor = 6;
long int Duration_High;
long int Pulse_Wait_Time = 3000000;

void setup() {  
Serial.begin(9600);  
pinMode(IR_Sensor,INPUT);
digitalWrite(IR_Sensor,HIGH);
}

void loop() {
  Duration_High = pulseIn(IR_Sensor, LOW, Pulse_Wait_Time ); //,Pulse_Wait_Time
Serial.println(Duration_High);
}

//**********************code end******************************//

Bartje64, 2015-02-25:
I've done some more experimenting with this using the Due. The duration should be around 40,000 Micros, yet on the Due it varies quite a bit and never gets any higher than 7-8000 micros.
This leads me to believe that perhaps the "on-state" voltage value is quite low?

facchinm, 2015-03-06:
Hi @Bartje64 , there seem to be no problem with the Due pulseIn() function.
Maybe you could try this sketch (https://gist.github.com/facchinm/5815f83b55b1ef86dc6b - connect pin13 and pin3) to check if your board works correctly. 

Your routine correctly sets pin6 as INPUT+PULLUP and it works as intended in both UNO and Due. The internal pullups are different (UNO: 20-50kΩ
DUE: 50-150kΩ) so maybe your capacitor is too big for the DUE setup.

closed 2015-03-16 by facchinm