arduino/Arduino #2690: 1.5.x, 1.6.x IDEs no longer include "make"

WestfW, 2015-02-25:
It used to be possible to use the avr toolchain included with the arduino IDE as a general purpose tool set for building programs outside of the arduino IDE.  In particular, the optiboot bootloader was set up so that it could be built by people who had ONLY the arduino IDE installed.
In 1.5 (and 1.6), the IDE install no longer includes the "make" program, so this no longer works :-(
(On windows, the entire set of "miscellaneous" tools from hardware/tools/avr/utils/bin is no longer present.)
(note that these tools are not avr-specific, so a new "generic" tools directory would suffice for use by both avr and sam things.)

ffissore, 2015-02-25:
Hi @WestfW 
Indeed `make` is not part of the toolchain any more.

It was part of the toolchain because we used one that others packaged and there were slight differences between win, mac and linux: since 1.5.7 we are using a toolchain we package, so we are sure they are all equal and we are able to patch them.

Since the IDE does not use `make`, it didn't make sense to include a tool only used by those using other IDEs. Better to give those IDEs control over which tool to use.

You can install `make` on your own (on Window using cygwin or mingw), or contact the other IDE author and ask for having `make` included, or (better) fork repo https://github.com/arduino/toolchain-avr, add `make` to the list and build it yourself

WestfW, 2015-02-26:
Ah.  I had assumed that 1.5.x was extracting files from the Atmel command line tools install or something (rather than winavr), rather than going all the way to a source-based scheme.  I guess there's no good reason for an Arduino install to include a complete set of CLI tools; but it was convenient.

closed 2015-02-27 by ffissore