arduino/Arduino #2691: IDE 1.6, Exception ...UnsatisfiedLinkError: jssc.SerialNativeInterface.getSerialPortNames() ...

MichaelJonker, 2015-02-25:
IDE 1.6 crashes when I have a Bluetooth mouse connected to the computer with:
Exception in thread "main" java.lang.UnsatisfiedLinkError: jssc.SerialNativeInterface.getSerialPortNames()[Ljava/lang/String;
        at jssc.SerialNativeInterface.getSerialPortNames(Native Method)
        at processing.app.SerialPortList.getWindowsPortNames(Unknown Source)
        at processing.app.SerialPortList.getPortNames(Unknown Source)
        at processing.app.SerialPortList.getPortNames(Unknown Source)
        at processing.app.Serial.list(Unknown Source)
        at cc.arduino.packages.discoverers.SerialDiscovery.discovery(Unknown Source)
        at cc.arduino.packages.DiscoveryManager.discovery(Unknown Source)
        at processing.app.Editor.populatePortMenu(Editor.java:998)
        at processing.app.Editor.buildToolsMenu(Editor.java:720)
        at processing.app.Editor.buildMenuBar(Editor.java:488)
        at processing.app.Editor.<init>(Editor.java:215)
        at processing.app.Base.handleOpen(Base.java:727)
        at processing.app.Base.handleOpen(Base.java:692)
        at processing.app.Base.handleNew(Base.java:595)
        at processing.app.Base.<init>(Base.java:297)
        at processing.app.Base.main(Base.java:166)

see also:
http://forum.arduino.cc/index.php?topic=299211.75
http://forum.arduino.cc/index.php?topic=299211.120

Bluetooth mouse used to be an issue already in IDE 1.0.5, see related postings and solution: 
http://forum.arduino.cc/index.php?topic=46977.0
http://forum.arduino.cc/index.php?topic=141797.0

rxtxSerial.dll was replaced with JSSC. Did it bring back a Bluetooth bug ?
Possible solution: (I have not try this yet, I also guess it could be refined)
https://alselectro.wordpress.com/2014/01/26/arduinosolution-to-launch4j-problem/

Michael

ffissore, 2015-02-25:
jssc is not affected by windows bluetooth bug. As for the UnsatisfiedLinkError, it affects a small number of windows users: we are going to try to solve it by upgrading the JVM shipped with the IDE, that's basically what alselectro suggests in his blog

MichaelJonker, 2015-03-02:
Hello,
tested with PR-2710-BUILD-198, yes it is now fully working.
Indeed (shame on me), I have been a bit premature in my conclusions that this bug could have be triggered by the presence of a bluetooth mouse... (as there used to be such an issue in 1.0.5).
However, the observations were a bit weird: the same laptop, the same installation, it worked in the office, not at home.
I now conclude that my firewall settings when at home were too restricted for the 1.6 version.
And if this was the case, then indeed it is a good thing that the updated version does not depend on unrestricted network access.

Thanks!

closed 2015-03-02 by MichaelJonker