arduino/Arduino #2704: Compiler does not recognize changes in libraries

moritz1671y, 2015-02-26:
While debugging a very simple library i made, i fixed an error in the library code (.cpp) but the compiler did not update the library and tried to use the precompiled file with the error still in it. Even after restarting the computer the precompiled file was still used. An option to clear the precompiled files and recompile everything would be useful.

ffissore, 2015-02-26:
Can you share the whole verbose output genereted by the IDE?

moritz1671y, 2015-02-26:
I had multiple errors. The main problem is that i am not able to tell the
IDE to recompile the library files.

ffissore, 2015-02-27:
That's clear. However without additional details we can't reproduce the
problem. First step is to share the whole verbose output, indicating the
library files that are not compiled

PaulStoffregen, 2015-03-02:
The IDE automatically recompiles library files that have changed.

moritz1671y, 2015-03-02:
In my case it didn't. Is that a recent addition to the IDE?

ffissore, 2015-03-02:
It is its default behaviour since the beginning (as far as I know). Did you manage to produce the verbose output I was asking for (file, preferences, verbose compile)? The ideal output is one generated right after having started the IDE, with you pointing at the row where your library is compiled and some code that shows that the IDE is using a previously compiled version

PaulStoffregen, 2015-03-02:
It started here, in late 2011.  Before this, Arduino always recompiled all files.

http://code.google.com/p/arduino/issues/detail?id=638

I believe Arduino 1.0.1 was the first to merge this feature.

moritz1671y, 2015-03-02:
@Frederico Fissore I have not worked with the IDE since i discovered the
issue but i will send you the requested data as soon as I get  to it again.

moritz1671y, 2015-03-02:
@Federico Fissore I am sorry for misspelling your name.

ffissore, 2015-03-02:
Not a problem. Can't wait to see that data

moritz1671y, 2015-03-02:
/Documents/Arduino/libraries/SSDcontrol/SSDcontrol.cpp:95:28: error: no 'void SSDcontrol::SSDreset()' member function declared in class 'SSDcontrol'
 void SSDcontrol::SSDreset () {
                                              ^

Line 95 in SSDcontrol.cpp: void SSDcontrol::SSDreset() {
in SSDcontrol.h I declared: void SSDreset(); as public

cmaglie, 2015-03-04:
Hi @moritz1671y

since there is a compile error surely there are no object files that the IDE can reuse.
From the few information you provided it looks like an error in your code instead.

If you are really sure that this is a bug in the IDE, you should post:

* the **full** content of SSDcontrol.cpp and SSDcontrol.h
* your **full** sketch 
* the **full** verbose compile output (remember to activate the "verbose compile" option from the preferences) 

This should be enough to allow us to reproduce the issue (remember: if we can't reproduce the issue we can't solve it!).

You can use http://gist.github.com to copy and paste if needed.

Thanks!

agdl, 2015-04-09:
Please reopen this if needed

closed 2015-04-09 by agdl