arduino/Arduino #2706: results of basic arithmetic operation is incorrect

muratgorece, 2015-02-26:
hi.
long i = 367*2015;
Serial.println(i);

18609

mikaelpatel, 2015-02-26:
Try 

long i = 367*2015L;
Serial.println(i);

Why will this give the correct answer?

muratgorece, 2015-02-26:
Serial.println(367*2015) wrong
is this normal?

matthijskooijman, 2015-02-26:
Yes, this is normal according to the C / C++ language rules. Did you see what @mikaelpatel posted? Notice the trailing "L" in his second number and his question (yes, we're inviting you to do a bit of research on your own to figure things out, instead of just "giving away" the entire solution).

muratgorece, 2015-02-26:
but it turns out the right results in the desktop program

matthijskooijman, 2015-02-26:
Right. This is because the Arduino and your desktop have a different *architecture* / instruction set. In particular, the standard number type, `int` is only 16-bits wide on AVR (the architecture most Arduinos use), but is 32-bits on x86 and (IIRC) also amd64.

Does that help?

muratgorece, 2015-02-26:
It was okay now.
thanks.

closed 2015-02-26 by muratgorece