arduino/Arduino #2711: Verbose output follows the window focus

mastrolinux, 2015-02-27:
When uploading a sketch or doing everything writing to the console inside the IDE, if someone changes window the output follow the focus. This make debugging almost impossible with part of the output split over multiple windows.

wayoda, 2015-02-27:
I can confirm that this bug exists, but   
> This make debugging almost impossible with part of the output split over multiple windows.

is a bit of an exaggeration I think.
A simple workaround is to wait until the operation finishes before switching to another window.

mastrolinux, 2015-02-27:
If your sketch takes a while to upload, especially on networking upload, that means you could wait up to 1  min before switching window. It is for sure not an high prio bug, still something good to have. It can confuse new users.

ffissore, 2015-02-27:
Duplicate of #2680

closed 2015-02-27 by ffissore