arduino/Arduino #2714: explicite including of libraries plus support of header files (relative or absulute path)

VogonJeltz, 2015-02-27:
hi,
I would like to have explicite including of libraries and support of header files (relative or absulute path) used in the current project instead of implicit loading all libs either if wishful or not.
e.g.,
#include <stdio.h>
#include <math.h>
#include <stdlib.h>
#include <SD.h>
#include <SPI.h>
#include <DueTimer.h>
#include "myproject.h"
#include "\arduino\tools\graphtools.h"
#include <string.h>

If a lib is not included, it should be left off.

VogonJeltz, 2015-02-27:
#include <stdio.h>
#include <math.h>
#include <stdlib.h>
#include <SD.h>
#include <SPI.h>
#include <DueTimer.h>
#include "myproject.h"
#include "\arduino\tools\graphtools.h"
#include <string.h>

ffissore, 2015-02-27:
Can you make a short sketch that shows the problem? We can then use it to try to implement the feature you requested

VogonJeltz, 2015-02-27:
Currently I don't have one because it didn't work so far so I abandoned this thing.

e.g., for using sprintf and sscanf  stdio.h would have to be #included (or it woudn't work). 
But IF they are needed and stdio.h would be #included - then  all ANSI standard features also should have to be provided (e.g., float formatting).

In case a nano user would not like to have this full feature but just some basic features he should have to #include a different smaller lib  - let's call it nanoio.h, feat. a  sprintf_nano derivative of smaller memory claims (this argument was often heard when talking about the "full featrures".

For .h and .c files I just don't see how to save or open such a file in the Editor window additionally in an additional tab window. 
Ican't write code and save it to "myHeader.h", 
and if my current dir contains 2 files, e.g.
SDfgets.ino
ANSItools.h
I can't even find the header file listed using the file_open menu.

And If I currently start Sketch I get a program "BareMinimum" pattern. If I designed it to be a header then I can't "save_as" by a name like myheader.h or ANSItools.

That's roughly what it is about.

ffissore, 2015-09-17:
With the latest nightly, you can include files from absolute locations, as long as gcc understands them. As for opening those files in the IDE: no, the IDE is not a generic c++ editor

closed 2015-09-17 by ffissore

VogonJeltz, 2015-09-17:
thank you, but my question was more about that the Arduino IDE MUST NOT automatically include libs which are not explicetely included by the individual programmer!#

So if someone don't   include
#include <stdio.h >
he must not have access to e.g., sprintf.

But if one will  #include <stdio.h > the he must get full access to all GCC stdio.h functions such as sprintf even supporting float formatting and file fprintf and fscanf, even for AVRs.