arduino/Arduino #2715: Improvement of stdio.h functions

VogonJeltz, 2015-02-27:
hi,
I would appreciate if the number of stdio.h funcrions for file access could be impoved.
For file access we basically just have (file).print, println, .write,  and just .read (by single bytes).
But fprintf and fscanf would be appreiciated, too.

E.g.,that's why  I just faked fgets() to something which mimics the ANSI function.
It's been programmed quick and dirty, but IMO it's already an improvement.
You may polish it up for your libs and add it for use by SD if you wish.

char * fgets ( char * str,  int32_t  num,  File * stream ) {
  int32_t i=0;
  
  strcpy(str, "");     // will return a nullstring in case of failure, not a NULL pointer!
  while ( stream->available() && (i<num)   ) {    
     str[i]=stream->read() ;
     if(str[i] == '\0') goto ok; // end of string: \0
     else 
     if(str[i] == '\n')   {      // when found a CR at end of string
         str[++i] = '\0';        // add an additional \0 char after \n (ANSI Standard)
         //str[i] = '\0';        // optional: override CR by a Null-Termination
         goto ok;
     }
     else  ++i;         
  }  
  ok: return(str);
}

I just am trying to mimic more functions using multiple variable parameter lists ( ... ) provided by
#include < stdarg.h>
#include < stdio.h>
e.g.,

char * fprintf(File * stream,  char fmtstr[], ... )

but ofcourse it would be better to access the "real" stdio.h functionality instead of a faked and limited one.

peabo7, 2015-02-27:
Your fgets doesn't do what real fgets does, so you should call it something else, if you use it at all.

There are several problems:

It has a buffer overrun because it doesn't pay attention to the num parameter.
It never detects end of file, because 0 is not the indicator for end of file (-1 is, which is why you need to read into a int16_t).
The handling of end of file returning NULL is quite specific, in the case where end of file happens before reading any characters.
The loop terminates prematurely if stream->available() returns 0.

Try this instead (not tested, you should test it).

char * fgets ( char * str, int32_t num, File * stream ) {
&nbsp;&nbsp;int32_t i = 0;
&nbsp;&nbsp;
&nbsp;&nbsp;while (i < (num - 1)) {      // got room for newline and terminating null?
&nbsp;&nbsp;&nbsp;&nbsp;int16_t ch = stream->read();
&nbsp;&nbsp;&nbsp;&nbsp;if (ch < 0)         // end of file
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;break;
&nbsp;&nbsp;&nbsp;&nbsp;str[i++] = ch;
&nbsp;&nbsp;&nbsp;&nbsp;if ('\n' == ch)     // end of line
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;break;
&nbsp;&nbsp;}
&nbsp;&nbsp;
&nbsp;&nbsp;if (i) {              // room in buffer for terminating null
&nbsp;&nbsp;&nbsp;&nbsp;str[i] = 0;
&nbsp;&nbsp;&nbsp;&nbsp;return str;
&nbsp;&nbsp;}
&nbsp;&nbsp;else
&nbsp;&nbsp;&nbsp;&nbsp;return NULL;        // buffer too small or immediate end of file
}

Peter Olson

VogonJeltz, 2015-02-28:
thank you for your input, I'm just a beginner in using C, and hitns a very much appreciated.

I have a question nevertheless:
You wrote 
"It has a buffer overrun because it doesn't pay attention to the num parameter."
but I had coded in  1 line
"while ( stream->available() && (i < num) ) "

so num had been notified IMO very well - is that not true?

VogonJeltz, 2015-02-28:
BTW,
although peabo7's function mimics the "reals" fgets() very well, this is just why I appreciated to have full stdio.h and stdlib.h support by Sketch instead of having to fake all these functions by one one with a risc of severe failures. So why having to invent the wheel from the scratch? It's all here, be gpp. (CMIIW)

matthijskooijman, 2015-02-28:
I'm again confused. It seems that `fgets` is already available through libc (which is available and enabled by default):

```
matthijs@grubby:~$ grep fgets /usr/lib/avr/include/ -r
/usr/lib/avr/include/stdio.h:extern char        *fgets(char *__str, int __size, FILE *__stream);
```

The only thing that is _not_ available, is default streams (stdin/stdout/stderr), since they make no sense on a microcontroller, or functions like `fopen` to get a `FILE*` pointing to a file, since there is no default filesystem.

I previously suggested, in another ticket, to use `FDEV_SETUP_STREAM` to create a custom `FILE*` that wraps `Serial` (or whatever Arduino `Stream` you want). Then you can just use the already available `fgets`, `fprintf` (and `printf` if you assign `stdout`).

Does this not satisfy your need? Do you perhaps not understand what I'm suggesting? Or are you of the opinion that such `FILE*`-creating code should be included with Arduino by default?

VogonJeltz, 2015-02-28:
2x yes ;)
1,st I do not understand your suggestion ;)
2nd, IMO SD files  should be impplemented to Sketch (SD files) to work just the same as files usually work  by C and the common stdio.h functions 8-)

PaulStoffregen, 2015-02-28:
You really seem to have missed the concept that Arduino creates its own simplified APIs.

Arduino is designed to be a system that makes electronics easier for novices and hobbyists.  These things are done differently from standard C for very good, very intentional reasons.

You're not the first person to come along and loudly insist Arduino should abandon its long-standing and very successful design, only because you personally want it to work the same way as other systems.  You're not the first person to say Arduino's way is non-standard or technically inferior.  Many others have come before you and made pretty much all these same arguments.

The obvious reality is Arduino is a very successful platform **because** is provides simplified APIs.

VogonJeltz, 2015-02-28:
I know that it IS different, but it's not the best way it is supposed tobe.
IMO Arduino is C and so it should work like C in the first line, 
probably providing some stuff which could make it (additionally) a little simpler here and there.

So first of all comes full C  compatibility (like SNSI C99 plus stdio.h plus stdlib.h,

and then (maybe) comes some additional stuff.

PaulStoffregen, 2015-02-28:
No, you do not understand.

Arduino is Arduino with uses C internally.  It should work like Arduino.

VogonJeltz, 2015-02-28:
here in my topic about Sketch enhancements it is about making Arduino work with stdio.h functionality like C does by standard.
So IMO (about which is the current topic)  Arduino should be provide standard C compatibility.

Additionally of course it may provide additional features which makes it easier here and there to handle by hobbiists.

opening files by fopen is standard, so it should work with Arduino.
fprintf is standard for writing data to file, so it should work foe Arduino.
scanf is standard in C to read data from files, so it should work with Arduino.

If one likes to have addtionally SD-(myFile).println, why not? but that's secondary.

PaulStoffregen, 2015-02-28:
No, it's is Arduino's primary API.  You simply do not understand Arduino.

VogonJeltz, 2015-02-28:
I understand Arduino, but my topic is about enhancing  Arduino by C compatibility.
I think this is the point which YOU don't understand ?

PaulStoffregen, 2015-02-28:
No, you obviously do not understand Arduino at all.

VogonJeltz, 2015-02-28:
well, maybe start reading at the TOP.

mbanzi, 2015-02-28:
@PaulStoffregen gets Arduino :) thanks Paul

mikaelpatel, 2015-02-28:
Is this a discussion about the Arduino API? I always thought the API was an attempt at OOP/C++. I would really like to discussion some of the constructs and the usage of classes. I do not see the elegance that @PaulStoffregen seems to see. What I see is something that would not give computer science students a degree let alone work in industry :) Let us use some standard industry metrics and evaluate the Arduino core and libraries before doing some of the claims above. @VogonJeltz gets software standards and reuse. 

Cheers!

peabo7, 2015-03-01:
> thank you for your input, I'm just a beginner in using C, and hitns a very
> much appreciated.
>
> I have a question nevertheless:
> You wrote
> "It has a buffer overrun because it doesn't pay attention to [...]

Aha, I thought there was something funny about that line.  The message displays
differently in email from the list than it does when looking at it on Github.
 In email it says:

  while ( stream->available() && (i str[i]=stream->read() ;

My apologies!  Now I know to read on Github before replying.  The list software
is sensitive to text that could be construed as HTML, and the "< num" resembles
a tag well enough to fool it.

Peter Olson

PaulStoffregen, 2015-03-01:
@mikaelpatel - I would not use the word "elegance" to describe Arduino.  Your Cosa system is the only hardware API I've ever seen that truly deserves to be called elegant.

Arduino is approachable and learn-able and usable for novices and absolute beginners.  For anyone who wishes to argue otherwise, I can't hear you over the sound of Arduino's incredible success.

VogonJeltz, 2015-03-01:
to make my intuition clear:
i love the Arduino IDE because it's simple and handx and allows a very quick start for beginners, and I appreciate the convenient API for programming! 
I myself had never started with MCUs if it had not been possible as it turned out by the Arduino Sketch IDE.

But there are 2 points which have to be taken into account:
1st, not every Arduino user stays at Uno or Nanos but probably will proceed to Mega and Due and (hopefully soon) Tre for larger projects by higher demands
2nd, for higher demands one would appreciate to be no longer restricted to the Arduino Sketch world of "not quite" and "not exactly" (as once one John Hansen,  the author of Lego Mindstorms IDEs apostrophied talking about his own NQC and NXC compilers) but  might be able to use additionally either powerful functions of the real word of C which is no longer "not quite" or "not exactly" any longer !

Indeed, being tied to IDEs like Eclipse and committed to makefile and explicite static  ar dynamic linking most Arduino beginners had become desperate and abandoned all attempts  long before achieving the first simple program to run (just like me trying embedded C on an ARM 9 Linux ) - 
but when the first steps are done, and bigger projects start arising (e.g., for me: neural nets, astar, MonteCarlo filters, fast Fourier Transformation, speech recognition, autonomous path finding in real worls environments) then more powerful programming facilities would be needed, even though still staying fine with the Arduino IDE!

BTW, that's exactly why I'm eagerly waiting for the Tre feat. Sketch instead having to use the BBB or RasPi by their  horrible IDEs instead).

Now do you see my point?

I may say : I  do understand Arduino, but I want some elaboted advanced features ADDITIONALLY.

Would theat be OK?

mikaelpatel, 2015-03-01:
@PaulStoffregen 
Arduino is a great success due to (I my opinion) the needs for cheap hardware for teaching and hobby projects. It is not really Arduino per se but the trend towards "nerds" becoming respectable and that technology is now back on the curriculum (at least here in Sweden). 

Arduino could be even more successful with component based hardware AND software. There are a lot of software professionals that are willing to give a helping hand. Cosa is a demonstration of applied theory and methodology from over 10 years of teaching Computer Science at University and over 20 years experience of large scale embedded systems projects at one of the worlds largest Telecom companies. Cosa is in no way a product - so I am cheating and only halfway there. 

We could discuss how to add support for larger projects and for those who have come past being beginners. In that case I think the discussion will actually end up in software standards and being able to share software across several platforms. Software development is the largest cost in any project.

Already with the Arduino Mega the lack of support for larger projects became obvious. The whole SPI discussion is a good example. And we are not really done with that as there is a level missing when we go for multi-tasking.   

Cheers!

VogonJeltz, 2015-03-01:
YES, defintely YES:
multitasking (i.e.: preemptive multitasking maybe something similar like featured by POSIX  pthread) really IS an another issue!!

BUT PLEASE NO COMPULSATORY OOP!

But for the first step, acc. to the topic of this thread:
some additional stdio.h functions additionally  for the awsome Arduino IDE !

PaulStoffregen, 2015-03-01:
@mikaelpatel Yes, I agree, much needs to be done to facilitate larger projects.  This particular thread probably isn't a very productive way to talk of such things.

VogonJeltz, 2015-03-01:
supporting standard C syntax  is NOT productive for larger projects ?? 
Are  you SERIOUS ??

mikaelpatel, 2015-03-01:
@VogonJeltz
Back to the original issue (sorry for the diversion). Did you see this https://github.com/greiman/SdFat/tree/master/SdFat?

Bill has actually taken the time to implement the ios standard (https://github.com/greiman/SdFat/blob/master/SdFat/ios.h) for SD as far as I can tell. But the necessary parts for the AVR GCC stdio is not very hard and I agree should be implemented. If it should be in the Arduino core might be another issue as that is the product management dimension. 

The current Serial/Print/etc interface in Arduino could be improved to put it mildly. Compare for instance Cosa UART with Arduino Serial. Why not include an IOBuffer class from the start? Ring buffers are always needed. And why not make that implement a device interface? 

Last @PaulStoffregen, we might need to bring up the issue on the developers group? I would at least like to start discussing the need for back-end interfaces so that device drivers for the different bus protocols can be improved (e.g. I2C, SPI, 1-Wire). The Cosa code is full of delegation to interface such as the IOStream::Device to achieve this structuring of code. This has started to scale nicely. And actually a C++ variant of the different device driver structures in Linux. 

Cheers!

VogonJeltz, 2015-03-01:
Mikael,
thank you for your input!
Unfortunately I'm not used to github information organization and so I don't actually see what you mean.

But meanwhile I have written my own additional function as a workaround which mimics fprintf() sufficiently and so it's a substitute FTM.
(Nevertheless, a whole support stdio.h would be still appreciated!)

//********************************************************************************​***********  
int fprintf_ ( File * stream, const char fmtstr[], ... ) {
   char      str[1024];
   va_list   arguments;
   int16_t   num;
    
   va_start( arguments, fmtstr);
   num = vsprintf(str, fmtstr, arguments);
   // #debug
   // Serial.println( "string to write to file="); Serial.println( str);    
   stream->print(str);
   va_end( arguments );

   return num;
}
//********************************************************************************​***********

peabo7, 2015-03-02:
You can get a little extra safety by changing

num = vsprintf(str, fmtstr, arguments);

to

num = vsnprintf(str, sizeof str, fmtstr, arguments);

Peter Olson

VogonJeltz, 2015-03-02:
thank you very much, I will do so!
Finally things like these will enable me writing portable code which I will be able to utilize even later on different platforms!
(I already have tons of code written in a "not-quite-C" dialect which is now completely useless to me for Arduino, and I swore I will never make this mistake again.
Better no code at all than non-compatible C-(un)like code!!
So just again to the devs:

please make SD files compatible to C files in order to utilize stdio.h 
WHICH IS FINALLY  ALREADY HERE IN THE ARDUINO SYSTEM!

mikaelpatel, 2015-03-02:
@VogonJeltz 
Did you see this? http://www.nongnu.org/avr-libc/user-manual/group__avr__stdio.html and http://www.nongnu.org/avr-libc/user-manual/group__avr__stdio.html#gaea2b6be92ead4673bc487b271b7227fb.

Checkout the examples on how to bind a device to your own FILE. It is a matter of a few lines of code to bind the Arduino Serial or SD to the AVR stdio FILE and start using that. 

Remember this is AVR GCC stuff and not Arduino. If you consider that to be wrong you have to go to that right forum. 

Cheers!

VogonJeltz, 2015-03-02:
no, sorry, as I already wrote:
I don't understand where to look and what to change. I don't get clear with the github maze.
Can you just write down the code line I have to write in order to make SD a FILE *  ?
Otherwise I would prefer it to be available ready, finished, ot of the box.

mikaelpatel, 2015-03-02:
That will not happen :)

VogonJeltz, 2015-03-02:
What will not happen ?

mikaelpatel, 2015-03-02:
"...it to be available ready, finished, out of the (Arduino-) box, for Mega, Due, PCduino, and the same code for the Tre..." will not happen any time soon :)

VogonJeltz, 2015-03-02:
But that's exactly what my (this) topic is about! What part didn't you understand about
" I appreciated to have full stdio.h and stdlib.h support by Sketch instead of having to fake all these functions by one self " ?
stdio.h is already here, nothing has to be faked!
why not a have files on SD which work like files on embedded ARM-Linux MCUs ?

mikaelpatel, 2015-03-02:
@VogonJeltz 
I do not represent Arduino hardware or software. I was just trying to help you use AVR stdio and how you can bind functions to that to solve your issue. This is as far as I will go. Hope you get this issue solved. Cheers!

VogonJeltz, 2015-03-02:
well, I don't want a substitute of faked functions. 
Some of theses functions I already faked by my own, but that doesn't make e.g. fprintf() work with SD files, it just makes a fprintf_ faked function work with SD cards.

I want the original stdio.h and stdlib.h to work with the files on Arduino SD cards just as if they were real standard C files.

Where is the problem?

PaulStoffregen, 2015-03-02:
The problem is with you, and your lack of understanding of what Arduino is.

VogonJeltz, 2015-03-02:
why are you offending and insulting Arduino users by unqualified verbal abuse?
who do you think you are?

PaulStoffregen, 2015-03-02:
Before you label me arrogant, perhaps you should consider who is demanding an extremely successful open source project, used by millions of people, must change its design to meet his desires?

Before you cast yourself as the victim of verbal abuse, perhaps you could re-read some of the very angry message you've written here?

But most importantly, perhaps you should try to understand Arduino's design and goals?

PaulStoffregen, 2015-03-02:
Regarding who is who, I have personally fixed dozens of bugs and contributed numerous features to Arduino over the last 6 years.

What have you done, other than this angry rant insisting Arduino change its APIs?

VogonJeltz, 2015-03-02:
let's start with you - are you working for Arduino company or are you just an arrogant nerd?
Or both ?

mikaelpatel, 2015-03-02:
@VogonJeltz 
It took me ten minutes to fix the issue by add the necessary lines to get SD working with the AVR GCC stdio.h function. Unfortunately I would never share that with you - ever. Please close this issue and read more Vogon poetry :) - you actually got your name right :)

VogonJeltz, 2015-03-02:
tell me please what you did to work SD with gpp stdio.h (I don't want it to work with a different lib, but with the original gpp lib)