arduino/Arduino #2718: GUI issues when doing fask clicks on verify or upload buttons

bperrybap, 2015-02-27:
Here is an issue I'm seeing with the IDE.
This is triggered by a bad mouse and/or potentially a mouse driver.
I've seen this issue with lots of other software not just the Arduino IDE.

Here is the scenario. The left button on the mouse is not making as clean connection
as it should. i.e. it is creating lots of switch bounce.
Also, for some silly reason neither the mouse itself or the mouse driver are properly
filtering this switch bounce and and are sending several left button up/down events.

The IDE doesn't properly handle/ignore/block multiple button presses.
It may be do to the way it is handling the UI and its internal threads.
So what happens is that the if you click on the "upload" button more than once fairly quickly
(which this bounce issue does). The first click starts the compile/build process then the second click also
tries to kick it off and you end up with a confusing output that can contain errors yet kind of still appear to work.
Depending on the timing, it can work and print errors that it didn't or fail with odd error messages.

I'm assuming this is a s/w issue in the IDE GUI code in that it is not properly mutexing
the build code thread when the Verify and Upload buttons are pressed and
is allowing secondary button presses to kick of the process while the build process from
the previous button press is still pending.

Yeah I know this is being triggered by a bad mouse but it
seems like this is something that should be fixed in the GUI.

--- bill

ffissore, 2015-03-02:
It looks related to #2574

bperrybap, 2015-03-02:
Yes it is same issue. I think this is better describing the general threading problem.
There is some internal threading that is not properly mutexing out other threads during these operations.
It doesn't necessarily always clean itself up.
On LInux, depending on the timing avrdude can start up multiple times
and then get hung leaving an avrdude stranded out there that has to be killed manually. 

The IDE should be using mutexes to block out future events before kicking off these types of operations
so that you never end up with more than one of these types of operations running at once.

Another issue that might be related is the if you have two sketches open, which creates a window
for each, and you start a build on one, if you move the mouse focus over to the other window,
the stutus/error output will move to that window vs remaining in the sketch window where
the build was started.

ffissore, 2015-09-08:
Merged into #3771

closed 2015-09-08 by ffissore