arduino/Arduino #2719: Installing the IDE to the sketchbook folder causes havoc

PaulStoffregen, 2015-02-27:
Mistakenly placing a complete copy of the Arduino IDE into the sketchbook folder wrecks causes all sorts of subtle and difficult to understand problems.  Because the sketchbook can have "hardware" and "libraries" folders, those folders from a copy of Arduino cause libraries to incorrectly override others, without the platform and board specific matching meant to be used on the IDE's copies.

Crazy or unlikely as this situation sounds, it turns out some users, especially on Windows, will intuitively rename the extracted folder "Arduino" and place it into their Documents folder.  Or if they have multiple copies of Arduino, they may use File > Preferences to set the sketchbook, based on a misunderstanding that doing so lets them switch between their many copies of Arduino in various locations on their computer.

The IDE should perform some sanity checks on the sketchbook folder at startup, so make sure it is not actually a complete copy of the IDE.  If it is, some sort of nicely worded dialog box should be shown, to explain the mis-configuration to the user.  If the IDE actually runs, many things do work, especially sketches that don't use libraries.  But sketches using libraries tend to fail in very difficult to understand ways, especially when libraries used have multiple copies in "libraries" and also in the various platform folders under hardware.

PaulStoffregen, 2015-02-27:
The File > Preferences dialog should validate the folder entered.  If it's a location that contains a complete copy of the IDE, some sort of error message should be shown, and the sketchbook location should not change.

Defragster, 2015-02-28:
This is very important that the software should protect itself.  I went from a running system to a broken system just by making my personal sketches in a subdirectory of my Arduino installation.  I didn't just point at that folder but the Arduino folder above it because it very kindly showed me other Sketch folders in that directory.  This was a very subtle change I made with know understand or warning of the consequences, and was very difficult to diagnose and correct and it rendered my development system compromised and unusable for days in my case.

This should be a very simple change to make as the folder is specified during editing.  Ideally perhaps the SKETCH folder would not be included in the path to general build libraries as this as noted circumvents the protection and control systems built into the build process.  

I have three systems I did this to and all of them fail in the same way.  I started doing this when I was juggling multiple installations of Arduino for various hardware venders.  Once I created this Sketchbook folder anything in that folder related specifically to that hardware platform so organizing those various platforms and the related sketches in a single folder outside Arduino didn't make sense.

PaulStoffregen, 2015-04-04:
Here's another user who decided to locate his sketchbook folder INSIDE the Arduino IDE folder.

http://forum.arduino.cc/index.php?topic=313236.msg2171749#msg2171749

The IDE has several assumptions built into its design.  At the very least, we have 3 locations that are assumed to never be the same or overlap (one within another).  Or at least I think they're not supposed to ever overlap.  Is that right?  Or are some overlap cases allowed?

1: IDE install path
2: sketchbook folder
3: preferences+packages folder

By default, these should not overlap.  But obviously some users do change them, and then blame Arduino when things go wrong.  Some of that blame isn't entirely misplaced, since the IDE doesn't check if the install locations meets the design assumptions.

PaulStoffregen, 2015-05-12:
See issue #3120 for what might be a very confusing problem related to this issue.

Real users do end up with their Arduino software configured in ways where very confusing problems happen.

ffissore, 2015-06-03:
Fixed. A warning (popup or console) is diplayed if the IDE is installed into a subfolder (or same folder) of either settings folder or sketchbook. Check occurs at start up time and when attempting to save preferences

commit bede6967 (2015-06-03): Installation folder check both at startup and when user attempts to change
sketchbook location. Fixes #2719
closed 2015-06-03 by ffissore