arduino/Arduino #2725: IDE Interface meltdown

beshur, 2015-03-02:
Hi.

I have a problem which can be seen in this video https://monosnap.com/file/ViHXeHnjlNbvvTfkKuxBuqWna6XYu9
From a sudden everything starts to be redrawn and overlap, including console and top menu, and that looks very strange.
I've installed 1.6.0 yesterday, but it did not help.
Windows 8.1.

Also I've checked the [issue 2719](https://github.com/arduino/Arduino/issues/2719) and found that I had multiple copies of `libraries` in my sketch directory
[image]

I tried renaming `libraries` in the sketch folder just to check, but it did not help either.

What am I doing wrong?

Thanks.

ffissore, 2015-03-03:
Did you use the installer or did you unpack the zip? If the latter, where did you unpack the IDE? Can you try using the installer?

beshur, 2015-03-03:
I used the installer, @ffissore.

ffissore, 2015-03-03:
Very strange. Never seen it before. Can you try running the nightly version? http://arduino.cc/en/Main/Software#nightly Should not fix the issue, but we changed something that impact the graphical part of the IDE

agdl, 2015-04-09:
Reopen this if needed

closed 2015-04-09 by agdl