arduino/Arduino #2730: ATtiny85 problem with newest version of Arduino software

MishaBravo, 2015-03-04:
Details in the discussion on Reddit:
http://www.reddit.com/r/arduino/comments/2xv530/attiny85_problem_with_newest_version_of_arduino/

ffissore, 2015-03-05:
Last comment on reddit links to the right place for receiving support on tiny https://code.google.com/p/arduino-tiny/

closed 2015-03-05 by ffissore

cmaglie, 2015-03-05:
For the record, the link to the arduino-tiny cores from reddit is
http://code.google.com/p/arduino-tiny/downloads/detail?name=arduino-tiny-0150-0020.zip
A quick check on the zip file shows that this should be a 1.6 compatible version: I didn't tried it BTW.

You can also try the cores from @damellis: https://github.com/damellis/attiny/
In the readme there is a link to the 1.6 version: https://github.com/damellis/attiny/tree/ide-1.6.x