arduino/Arduino #2733: Missing documentation in MOST examples

GEORGESLAVRA, 2015-03-05:
Purchased several modules ( transceivers; accelerometers; displays and more) and tried to find tutorials and examples to use with the new modules. In almost every sketch provided, components attached are missing pin assignment  information is missing. The function that is derived ( sense output; analog data etc. ) is used but never declared as to what pin it comes from. Missing also are required "#include ??" statements in  the header block. There are some very well documented sketches and tutorials that enabled rapid learning and production of satisfying demonstrations. Sources of crippled sketches, etc. is consistent with all third parties along with official Arduino sites. It must be a tremendous disappointment for people just starting to learn computers and programming. 
Another downer is missing Macro info that decodes "C" codes to actual machine codes. I feel that a programmer should "know" exactly what functions are being carried out by the computer. Assembler language and absolute coding is knowledge is important for programmers. Much satisfaction in the completion of an application of coding and hardware integration would be appreciated by the "newbee" aspirant.

PaulStoffregen, 2015-03-05:
If you want anyone to do anything, you've going to have to be MUCH more specific.

At the very least, you'll need to list each deficient example (ideally with a link to its web page) with a brief explanation of exactly what's wrong with about each individual example.

agdl, 2015-04-09:
Please reopen this if needed

closed 2015-04-09 by agdl