arduino/Arduino #2735: AVR + analogWrite = glitch

Coding-Badly, 2015-03-06:
http://forum.arduino.cc/index.php?topic=303922.0

By using digitalWrite for analogWrite values of zero and 255 the AVR core creates a situation that can cause a glitch.

The solution is simple.  Do not treat zero and 255 as special cases.  digitalWrite disables PWM so the user can call digitalWrite in the rare circumstances when they want PWM turned off.

The only side-effect is with Fast PWM pins.  Instead of a clean "off" there will be a narrow spike for zero.

T-Kuhn, 2016-02-15:
Wow. Thanks for the Hint!
I changed the PWM resolution from 8 to 10 bits (via the TCCR4A & TCCR4B registers) and it worked beautifully (new scale 0 ~ 1024). Except that there was a strange peak at - yes, you guessed correctly - 255; Because that Number is for some reason treated as special case.

Since I don't see the reason behind the special treatment I would like to see the issue fixed.

peabo7, 2016-02-15:
You would only know if you had previously read the library source code.  (My condolences for your loss of time.)  A general recommendation would be to rely on the datasheet and also the library source code before making a change.

As for why it was implemented this way, I suspect some awkward treatment of the edge cases for the original 8 bit PWM.

Peter Olson

T-Kuhn, 2016-02-16:
Hello Peter Olson,
You are completely right. I also am aware of the fact, that my case (making the 8bit PWM 10bit and THEN experiencing some kind of malfunction of analogWrite) is very different from what Coding-Badly is talking about.

So - to get back to the actual topic - there might really be a problem with the way the function falls back to digitalWrite in the cases of 0 and 255. You can read Coding-Badly's findings here:
https://github.com/Coding-Badly/arduino-tiny/issues/128

elektor-labs, 2016-09-26:
I vote for the removal of at least the 255 check. Not only is the 255 check useless, it is also strange because the argument of analogWrite is an int and not an unit8_t and it hinders people wanting to use analogWrite with a higher resolution PWM signal from 16-bit timers.

The zero check is more involved due to the fact that timer0 is used in Fast PWM mode and all the others in Phase Correct PWM mode, which is why pins 5 & 6 behave differently at 0% duty-cycle compared to pins 3, 9, 10 & 11. (On the other hand, they also run at a higher frequency, so they might as well be treated as a different kind of PWM signals.)

To get past the zero check one could (for timer0 only) for the case of 0% invert the PWM and set it to 100%, like this:
`if (val==0) { val=255; sbi(TCCR0A, COM0A0); } else cbi(TCCR0A, COM0A0);`
right after the `case TIMER0A:` & `case TIMER0B:`
That will produce real 0% and 100% PWM on pins 5 & 6 without having to fiddle with the pin modes.