arduino/Arduino #2737: Compiler errors don't highlight corresponding line of code.

damellis, 2015-03-06:
In past Arduino versions, when compiling a sketch the line of code corresponding to the first error would be highlighted. This was a big help in assisting users in locating (and fixing) the error, without necessarily needing to read through the (often-confusing) error message in detail. At some point, this behavior seems to have stopped. For example, I don't see it in the latest master branch (1.6.1) on Mac OS X (10.10.2). It would be great to bring back this functionality.

ffissore, 2015-03-13:
Report back here once fixed http://forum.arduino.cc/index.php?topic=307531.0

ffissore, 2015-04-10:
Fixed nightly available in the next few hours

closed 2015-04-10 by ffissore

damellis, 2015-04-10:
Great!

commit 4c9e5fc5 (2015-04-10): Restored error messages. Got rid of MessageSyphon as ther were losing some error messages. Fixes #2737