arduino/Arduino #2740: Wifi Shield Firmware Update

golfguy777, 2015-03-09:
I have been trying to update the firmware on my wifi shield for the past two weeks. However, the link that contains the firmware code on github gives me a 404 error. Would someone be able to help me with this? I've had no luck with this issue.

CaptCoonoor, 2015-03-09:
Can you give us link, so that we can help you?

golfguy777, 2015-03-12:
Hello,

Apologies for the late reply. The link with the incorrect instructions to
install the new firmware is here:
http://arduino.cc/en/Hacking/WiFiShieldFirmwareUpgrading,
and the github link for the code to install is here:
https://github.com/arduino/Arduino/tree/master/hardware/arduino/firmwares

The URL with the correct firmware code is:
https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/firmwares/wifishield

Regards,
Jeff

agdl, 2015-06-05:
like in #2896

closed 2015-06-05 by agdl