arduino/Arduino #2744: vsscanf compiles for Due but not for Mega

VogonJeltz, 2015-03-10:
calling

vsscanf ( )

it compiles and runs fine  for my Due - but not for my Mega !

included libraries:

#include < stdio.h>
#include < stdarg.h>
#include < SPI.h>
#include < SD.h>

    Quote:
    Build options changed, rebuilding all
    SDfprintf_fscanf101.ino: In function 'int32_t fscanf_(File*, const char*, ...)':
    SDfprintf_fscanf101.ino:46:38: error: 'vsscanf' was not declared in this scope
    Fehler beim Übersetzen

matthijskooijman, 2015-03-10:
This seems to be a limitation of avr-libc, which for some reason doesn't implement vsscanf. E.g. see http://forum.arduino.cc/index.php?topic=273201.0

If you want to see this fixed, you should file a b ug report with avr-libc. Adding this function to the Arduino core is really not the right place, especially since it has no place in the official API.

If you just want to get things working now, it seems there are a few implementations floating around the net that you can add to your own code if you google for "avr vsscanf".

closed 2015-03-10 by matthijskooijman

VogonJeltz, 2015-03-10:
no, I want it actually to be included in the standard Atrduino IDE and API.
It makes no sense IMO that identical code with identical standard function calls work for the Due and not for the AVRs ! 
I need the whole functionality by all included standard libs both on my Megas and on my Dues !

matthijskooijman, 2015-03-10:
Sorry, I don't have time to keep having the same discussion with you. Good luck with your projects.

VogonJeltz, 2015-03-10:
tbh, actually not a very innovative place  and atmosphere over here.

PaulStoffregen, 2015-03-10:
Maybe you should switch to mbed?

VogonJeltz, 2015-03-11:
no I just use the Arduino Sketch IDE and I just appriciated it to be even more perfect.

BTW, this saying was already told to us students demonstrating in the late 60ies for Democracy, anti-Rassism and againstthe  Vietnam war:
"If you don't like it here - why don't you go to the Russian Occupation Zone ?