arduino/Arduino #2745: vsscanf() does not compile for my Mega, although the code compiles and runs fine for my Due

VogonJeltz, 2015-03-10:
(previous post disappeared..?)

vsscanf() does not compile for my Mega, although the code compiles and runs fine for my Due:
same code in both cases !

stdio.h and stdarg.h are always #included.

error msg for Mega:

Build options changed, rebuilding all
SDfprintf_fscanf101.ino: In function 'int32_t fscanf_(File*, const char*, ...)':
SDfprintf_fscanf101.ino:46:38: error: 'vsscanf' was not declared in this scope
Fehler beim Übersetzen

no error for Due though.

matthijskooijman, 2015-03-10:
You previous post is still here:  #2744

However, I closed it, because it's not something that's up to Arduino to fix. I'm also closing this one, since it duplicates  #2744.

closed 2015-03-10 by matthijskooijman

VogonJeltz, 2015-03-10:
I thought this github thing here is for Arduino ?

matthijskooijman, 2015-03-10:
Yes, it is. However, the problem you describe is a bug / omission in avr-libc, which is another project _used_ by Arduino. To fix this, avr-libc needs to be fixed, so I suggest you file a bug report with that project. See http://savannah.nongnu.org/bugs/?group=avr-libc and http://www.nongnu.org/avr-libc/bugs.html.

VogonJeltz, 2015-03-11:
ok, sorry I just use Arduino, I don't know anything so far about  avr-libc or savannah.nongnu. I'm using Arduino just since a couple of months. So I just know this forum or how it might be called to report about Arduino issues.