arduino/Arduino #2746: Compilation error

tailleurp, 2015-03-10:
Compiled code truncated,  starting IDE > 1.5.5.  as result, execution fails.

IDE 1.5.5 : Le croquis utilise 48 838 octets (18%) de l'espace de stockage de programmes.Le maximum est de 258 048 octets.
Les variables globales utilisent 4 014 octets (48%) de mémoire dynamique, ce qui laisse 4 178 octets pour les variables locales. Le maximum est de 8 192 octets.

IDE >1.5.5 (1.6.1) : Le croquis utilise 44 690 octets (17%) de l'espace de stockage de programmes.Le maximum est de 253 952 octets.
Les variables globales utilisent 3 770 octets (46%) de mémoire dynamique, ce qui laisse 4 422 octets pour les variables locales. Le maximum est de 8 192 octets.

Possible issue : repeated #if  clause in single code ignored
Sample code
/////////////////////////////////////////////////////
// DIY alarm : RFID module
/////////////////////////////////////////////////////
//Variable
int reader = READER;
int creader = 0;

//define badge reader
#if READER > 0 
  PN532_HSU pn532hsu1(Serial1);
  PN532 nfc1(pn532hsu1);
#endif
#if READER > 1
  PN532_HSU pn532hsu2(Serial2);
  PN532 nfc2(pn532hsu2);
#endif
#if READER > 2
  PN532_HSU pn532hsu3(Serial3);
  PN532 nfc3(pn532hsu3);
#endif

///////////
// Init NFC
///////////
void initnfc() {
  char chip[15],firm[14];
  uint32_t versiondata;
  
#if READER >0
  nfc1.begin();    // init RFID on Serial 1
  versiondata = nfc1.getFirmwareVersion();
  if (! versiondata) {
    logserial("0,0","Didn't find PN53x board","on","Serial1");
    while (1); // halt
  }
  // Got ok data, print it out!
  sprintf(chip, "NFC chip PN5%x",(versiondata>>24) & 0xFF);
  sprintf(firm, "Firmware v%d" ,(versiondata>>16) & 0xFF );
  sprintf(firm, "%s.%d" ,firm, (versiondata>>8) & 0xFF );
  // configure board to read RFID tags
  nfc1.SAMConfig(); 
  logserial("0,0",chip,firm,"Ready on Serial1");
#endif
#if READER >1
  nfc2.begin();    // init RFID on Serial 2
  versiondata = nfc2.getFirmwareVersion();
  if (! versiondata) {
    logserial("0,0","Didn't find PN53x board","on","Serial2");
    while (1); // halt
  }
  // Got ok data, print it out!
  sprintf(chip, "NFC chip PN5%x",(versiondata>>24) & 0xFF);
  sprintf(firm, "Firmware v%d" ,(versiondata>>16) & 0xFF );
  sprintf(firm, "%s.%d" ,firm, (versiondata>>8) & 0xFF );
  // configure board to read RFID tags
  nfc2.SAMConfig();
  
  logserial("0,0",chip,firm,"Ready on Serial2");
#endif
#if READER >2
  nfc3.begin();    // init RFID on Serial 3
  versiondata = nfc3.getFirmwareVersion();
  if (! versiondata) {
    logserial("0,0","Didn't find PN53x board","on","Serial3");
    while (1); // halt
  }
  // Got ok data, print it out!
  sprintf(chip, "NFC chip PN5%x",(versiondata>>24) & 0xFF);
  sprintf(firm, "Firmware v%d" ,(versiondata>>16) & 0xFF );
  sprintf(firm, "%s.%d" ,firm, (versiondata>>8) & 0xFF );
  // configure board to read RFID tags
  nfc3.SAMConfig();
  
  logserial("0,0",chip,firm,"Ready on Serial3");
#endif
}

/////////////
// read badge
/////////////
void readbadge() {
#if READER > 0
  uint8_t success;
  uint8_t uid[] = { 0, 0, 0, 0, 0, 0, 0 };  // Buffer to store the returned UID
  uint8_t uidLength;                        // Length of the UID (4 or 7 bytes depending on ISO14443A card type)
  char badge[9];                            // store the badge uuid
  // Wait for an ISO14443A type cards (Mifare, etc.).  When one is found
  // 'uid' will be populated with the UID, and uidLength will indicate
  // if the uid is 4 bytes (Mifare Classic) or 7 bytes (Mifare Ultralight)
  creader += 1 ;
  creader = creader % reader ;
//  Serial.println(creader, DEC) ;
//  long bstart = millis() ;
  switch (creader) {
    case 0:
      success = nfc1.readPassiveTargetID(PN532_MIFARE_ISO14443A, uid, &uidLength);
      break;
#if READER > 1
    case 1:
      success = nfc2.readPassiveTargetID(PN532_MIFARE_ISO14443A, uid, &uidLength);
      break;
#endif
#if READER > 2
    case 2:
      success = nfc3.readPassiveTargetID(PN532_MIFARE_ISO14443A, uid, &uidLength);
      break;
#endif
  }
//  long bend = millis() ;
//  Serial.println(bend-bstart, DEC) ;
  if (success) {  
    if (uidLength == 4)
    {
      // We probably have a Mifare Classic card ... 
      sprintf(badge, "%X%X%X%X", uid[0], uid[1], uid[2], uid[3]);
      uint8_t keya[6] = { 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF };
	  
	  // Start with block 4 (the first block of sector 1) since sector 0
	  // contains the manufacturer data and it's probably better just
	  // to leave it alone unless you know what you're doing
      switch (creader) {
        case 0:
          success = nfc1.mifareclassic_AuthenticateBlock(uid, uidLength, 4, 0, keya);
          break;
#if READER > 1
        case 1:
          success = nfc2.mifareclassic_AuthenticateBlock(uid, uidLength, 4, 0, keya);
          break;
#endif
#if READER > 2
        case 3:
          success = nfc3.mifareclassic_AuthenticateBlock(uid, uidLength, 4, 0, keya);
          break;
#endif
      }  
      if (success)
      {
        if (disp_rfid) {
          logserial("3,2","RFID Card",badge,"Authenticated");
          disp_rfid=!disp_rfid;
        }
        else
        {
          logserial("3,2","RFID Card","","Authenticated"); 
          checkbadge(badge);
        }
      }
      else
      {
        logserial("3,3","RFID Card","","authentication failed");
      }
    }
  }
#endif
}

//////////////
// Check Badge
//////////////
void checkbadge(char badge[]){
  badge_count += 1;
  security tuser;
  badge_start = millis() ;
  bool badgeok = false;
  // compare to rfid in EEPROM
  secs_count = EEPROM.readByte(e_addr_secs);  
  for(byte i=0; i < secs_count; i++) {
    addr = e_addr_secs + 1 + i * sizeof(tuser);
    EEPROM.readBlock(addr, tuser);
    if(strcmp(badge, tuser.rfid) == 0) {
      //One rfid match 
      badge_count = 0;      
      badgeok = true;     
      rfidok(tuser.user);
      break;
    }
  }
  // badge not found
  if (! badgeok) {
    disp(iteminlist(bmsg,0),"");
    logserial("3,0",iteminlist(bmsg,0),"","");
    bipbad();
    if(badge_count >= 3) {
      alarmpwd(4);
    }
    delay(1000);
  }
}

////////////////
// RFID Badge OK
////////////////
void rfidok(char user[]) {
  disp(iteminlist(bmsg,1),user);
  logserial("3,1",iteminlist(bmsg,1),user," ");
  bipgood();
  delay(1000);
  if(activated > 0 ) {
    deactivate();
  } else {
    activate(2);
  }
}

///////////////////////////
// reset badge read counter
///////////////////////////
void resetbadgecount() {
  badge_count = 0;
  badge_start = 0;
  pwd_cancel = 0;
}

cmaglie, 2015-03-11:
The sketch does not compile:

```
     [exec] sketch_mar11a.ino:5:14: error: 'READER' was not declared in this scope
```

please post the full sketch, what is the expected result and what you actually get.

tailleurp, 2015-03-11:
Hello,

 

The full sketch in enclosed files.

Main is alarm8.ino, all to be in alarm8 directory

You will need libraries: DS1307new, EEPROMEx, Keypad, PN532, PN532_HSU, SerialCommand

SerialCommand : you need to edit SerialCommand.h, change to have #define MAXSERIALCOMMANDS    16

 

will not be easy for you to see result.

This code for a MEGA2560 board running a house alarm system with some devices connected : PIR and Hall sensors, LCD, Keypad and RFID readers.

The code manage to be configurable: define constant READER to setup 1 to 3 connected RFID readers to serial 1 to 3, depending what is installed ( see CUSTOMISATION AREA  in alarm8.ino). So, depending this constant, with #if READER directives in alarm_rfid.ino, we compile to drive 1 or 2 or 3 RFID readers

 

with IDE1.5.5, al works fine.

with higher IDE,  code compile without errors,  but result is shorter, and at the end doesn't run properly, RFID readers doesn't work, code stop.

 

Regards

> Message du 11/03/15 17:12
> De : "Cristian Maglie" 
> A : "arduino/Arduino" 
> Copie à : "tailleurp" 
> Objet : Re: [Arduino] Compilation error (#2746)
> 
>
> The sketch does not compile:

[exec] sketch_mar11a.ino:5:14: error: 'READER' was not declared in this scope

cmaglie, 2015-03-11:
I cannot see any file, I guess that github filtere all the attachments.

Before going further, maybe you want to check if the compiler is getting the `#ifdef` right: the fact that the code size is reduced could mean that all the code inside ifdef is being excluded.

May you try manually change the ifdefs to `#ifdef 0` or `#ifdef 1' where needed and see if it works?

tailleurp, 2015-03-12:
> Message du 11/03/15 22:11

Yes, looks like attachements filtered.
Honnestly, I'm not blocked by the issue ( compile with IDE1.5.5 works ). My intention was only to report it for improvement for futur versions.
I will try to get ifdef out and let you know later on.

ffissore, 2015-03-12:
@tailleurp the reduced size of the sketch can be an effect of the updated compiler bundled with the IDE (1.5.5 shipped gcc 4.3.2, 1.6.1 ships version 4.8.1)

As for the #if you may want to give a spin to the IDE downloadable here https://github.com/arduino/Arduino/pull/2729 (bottom of the page)
It bundles a new preprocessor and produces a much more correct code

ffissore, 2015-03-12:
also, thanks for your report ;)

ffissore, 2015-03-12:
@tailleurp if you prefer, you can put your sketch code on gist: https://gist.github.com/. It's a tool for sharing source code, perfect for your case

tailleurp, 2015-03-12:
> Message du 12/03/15 08:50

Hello,

 

here it is https://gist.github.com/tailleurp/8744af7bf672c86e4e35

 

Regards

ffissore, 2015-09-11:
Fixed by #3779

closed 2015-09-11 by ffissore