arduino/Arduino #2747: section type conflict

sverrisson, 2015-03-10:
Arduino: 1.6.1 (Mac OS X), Board: "Arduino Micro"

Build options changed, rebuilding all
/Users/XX/Documents/Arduino/libraries/RBL_nRF8001/RBL_nRF8001.cpp:25:23: error: setup_msgs causes a section type conflict with __c
 static hal_aci_data_t setup_msgs[NB_SETUP_MESSAGES] PROGMEM = SETUP_MESSAGES_CONTENT;
                       ^
In file included from /Users/XX/Documents/Arduino/libraries/BLE/hal_platform.h:32:0,
                 from /Users/XX/Documents/Arduino/libraries/BLE/lib_aci.h:34,
                 from /Users/XX/Documents/Arduino/libraries/RBL_nRF8001/RBL_nRF8001.h:17,
                 from /Users/XX/Documents/Arduino/libraries/RBL_nRF8001/RBL_nRF8001.cpp:14:
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WString.h:38:74: note: '__c' was declared here
 #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
                                                                          ^
/Users/XX/Documents/Arduino/libraries/RBL_nRF8001/RBL_nRF8001.cpp:368:32: note: in expansion of macro 'F'
                 Serial.println(F("Advertising started"));
                                ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

cmaglie, 2015-03-11:
Where I can find the library TBL_nRF8001 to check this?

sverrisson, 2015-03-11:
Hi,

Here is a link to the libraries used: http://redbearlab.com/getting-started-blendmicro <http://redbearlab.com/getting-started-blendmicro>

Under software checklist.

Here is the code I was trying to compile:

I tried with older arduino version 1.0.5 and that compiles.

Regards, Hannes

cmaglie, 2015-03-11:
It seems that redbear lab has release some fix upstream, but not yet updated the guide, try this ones:

Latest ble-SDK library (0.9.7.beta):
https://github.com/NordicSemiconductor/ble-sdk-arduino/archive/0.9.7.beta.zip

Latest nRF8001 (not yet tagged with a version, you should try the master):
https://github.com/RedBearLab/nRF8001/archive/master.zip

sverrisson, 2015-03-11:
Arduino: 1.6.1 (Windows 8.1), Board: "Arduino Yún"

C:\Users\XX\Documents\Arduino\libraries\BLE\bootloader_setup.cpp:2:20: fatal error: EEPROM.h: No such file or directory

 #include <EEPROM.h>

                    ^

compilation terminated.

Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

----- Upprunalegt skeyti -----
Frá: Cristian Maglie <notifications@github.com>
Til: arduino/Arduino <Arduino@noreply.github.com>
Cc: sverrisson <hannessv@simnet.is>
Sent: Wed, 11 Mar 2015 20:57:05 -0000 (GMT)
Efni: Re: [Arduino] section type conflict (#2747)

It seems that redbear lab has release some fix upstream, but not yet updated the guide, try this ones:

Latest ble-SDK library (0.9.7.beta):
https://github.com/NordicSemiconductor/ble-sdk-arduino/archive/0.9.7.beta.zip

Latest nRF8001 (not yet tagged with a version, you should try the master):
https://github.com/RedBearLab/nRF8001/archive/master.zip

---
https://github.com/arduino/Arduino/issues/2747#issuecomment-78369698

cmaglie, 2015-03-11:
add `#include <EEPROM.h>` at the beginning of your sketch

cmaglie, 2015-07-01:
Closing for lack of feedback, please reopen if the problem persists.

closed 2015-07-01 by cmaglie

boblansdorp, 2015-08-18:
I have the same problem.
Arduino: 1.6.5 (Windows 7), Board: "Blend Micro 3.3V/8MHz"
....

....\Documents\Arduino\libraries\BLE_SDK_for_Arduino\src\bootloader_setup.cpp:2:20: fatal error: EEPROM.h: No such file or directory
 #include <EEPROM.h>
                    ^
compilation terminated.

...

matthijskooijman, 2015-08-19:
@cmaglie already showed the solution for this:

> add #include <EEPROM.h> at the beginning of your sketch