arduino/Arduino #2752: 1.6.1 Sketches fail to link with R_AVR_13_PCREL due to interrupt service routine too far for startup rjmp in vector

ryanneve, 2015-03-11:
This seems to be a duplicate of  [#1071](https://github.com/arduino/Arduino/issues/1071) and [#1901](https://github.com/arduino/Arduino/issues/1901). Both of which are marked closed, but I'm experiencing this on 1.5.8, 1.6, 1.6.1 and the nightly as of 2015/03/11

I have two separate projects which are now getting this error when I try to compile. Both projects are big, run on a mega2560 and use PROGMEM and F() extensively. It was working this morning, but I must have added another F() and now it doesn't.
One of these projects will compile fine on 1.0.6. The projects are too big to post here, but I can provide a git link if you need it.
```
Using library music in folder: D:\Development\Arduino\libraries\music (legacy)
Using library Logger_SD in folder: D:\Development\Arduino\libraries\Logger_SD (legacy)
Using library FSM in folder: D:\Development\Arduino\libraries\FSM (legacy)
Using library SdFat in folder: D:\Development\Arduino\libraries\SdFat (legacy)
Using library LiquidCrystal in folder: D:\Development\Arduino\libraries\LiquidCrystal (legacy)
Using library SPI in folder: C:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\SPI 
Using library EEPROM in folder: C:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\EEPROM 
Using library RTClib in folder: D:\Development\Arduino\libraries\RTClib (legacy)
Using library Wire in folder: C:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\Wire 

C:\Arduino\arduino-nightly/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Arduino\arduino-nightly\hardware\arduino\avr\cores\arduino -IC:\Arduino\arduino-nightly\hardware\arduino\avr\variants\mega -ID:\Development\Arduino\libraries\music -ID:\Development\Arduino\libraries\Logger_SD -ID:\Development\Arduino\libraries\FSM -ID:\Development\Arduino\libraries\SdFat -ID:\Development\Arduino\libraries\LiquidCrystal -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\SPI -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\EEPROM -ID:\Development\Arduino\libraries\RTClib -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\Wire C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoEE.cpp -o C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoEE.cpp.o 
C:\Arduino\arduino-nightly/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Arduino\arduino-nightly\hardware\arduino\avr\cores\arduino -IC:\Arduino\arduino-nightly\hardware\arduino\avr\variants\mega -ID:\Development\Arduino\libraries\music -ID:\Development\Arduino\libraries\Logger_SD -ID:\Development\Arduino\libraries\FSM -ID:\Development\Arduino\libraries\SdFat -ID:\Development\Arduino\libraries\LiquidCrystal -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\SPI -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\EEPROM -ID:\Development\Arduino\libraries\RTClib -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\Wire C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoLCD.cpp -o C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoLCD.cpp.o 
C:\Arduino\arduino-nightly/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Arduino\arduino-nightly\hardware\arduino\avr\cores\arduino -IC:\Arduino\arduino-nightly\hardware\arduino\avr\variants\mega -ID:\Development\Arduino\libraries\music -ID:\Development\Arduino\libraries\Logger_SD -ID:\Development\Arduino\libraries\FSM -ID:\Development\Arduino\libraries\SdFat -ID:\Development\Arduino\libraries\LiquidCrystal -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\SPI -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\EEPROM -ID:\Development\Arduino\libraries\RTClib -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\Wire C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoPins.cpp -o C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoPins.cpp.o 
C:\Arduino\arduino-nightly/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Arduino\arduino-nightly\hardware\arduino\avr\cores\arduino -IC:\Arduino\arduino-nightly\hardware\arduino\avr\variants\mega -ID:\Development\Arduino\libraries\music -ID:\Development\Arduino\libraries\Logger_SD -ID:\Development\Arduino\libraries\FSM -ID:\Development\Arduino\libraries\SdFat -ID:\Development\Arduino\libraries\LiquidCrystal -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\SPI -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\EEPROM -ID:\Development\Arduino\libraries\RTClib -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\Wire C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoSD.cpp -o C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoSD.cpp.o 
C:\Arduino\arduino-nightly/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Arduino\arduino-nightly\hardware\arduino\avr\cores\arduino -IC:\Arduino\arduino-nightly\hardware\arduino\avr\variants\mega -ID:\Development\Arduino\libraries\music -ID:\Development\Arduino\libraries\Logger_SD -ID:\Development\Arduino\libraries\FSM -ID:\Development\Arduino\libraries\SdFat -ID:\Development\Arduino\libraries\LiquidCrystal -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\SPI -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\EEPROM -ID:\Development\Arduino\libraries\RTClib -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\Wire C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoStates.cpp -o C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoStates.cpp.o 
C:\Arduino\arduino-nightly/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Arduino\arduino-nightly\hardware\arduino\avr\cores\arduino -IC:\Arduino\arduino-nightly\hardware\arduino\avr\variants\mega -ID:\Development\Arduino\libraries\music -ID:\Development\Arduino\libraries\Logger_SD -ID:\Development\Arduino\libraries\FSM -ID:\Development\Arduino\libraries\SdFat -ID:\Development\Arduino\libraries\LiquidCrystal -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\SPI -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\EEPROM -ID:\Development\Arduino\libraries\RTClib -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\Wire C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoStates_collect.cpp -o C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoStates_collect.cpp.o 
C:\Arduino\arduino-nightly/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Arduino\arduino-nightly\hardware\arduino\avr\cores\arduino -IC:\Arduino\arduino-nightly\hardware\arduino\avr\variants\mega -ID:\Development\Arduino\libraries\music -ID:\Development\Arduino\libraries\Logger_SD -ID:\Development\Arduino\libraries\FSM -ID:\Development\Arduino\libraries\SdFat -ID:\Development\Arduino\libraries\LiquidCrystal -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\SPI -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\EEPROM -ID:\Development\Arduino\libraries\RTClib -IC:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\Wire C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoStates_diag.cpp -o C:\USERS\NEVE\APPDATA\LOCAL\TEMP\build4819664917475923097.tmp\LimnoStates_diag.cpp.o 
[... cut ...] L\TEMP\build4819664917475923097.tmp -lm 
c:/arduino/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6\libc.a(isspace.o): In function `isspace':
(.text.avr-libc+0x2): relocation truncated to fit: R_AVR_13_PCREL against symbol `__ctype_isfalse' defined in .text.avr-libc section in c:/arduino/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6\libc.a(cty_isfalse.o)
collect2.exe: error: ld returned 1 exit status
Error compiling.
```

arjenhiemstra, 2015-03-22:
Same here, quite big project (around 100k) and use PROGMEM and F() extensively.
No fix yet tho :(

Error:
/Applications/Arduino 1.6.app/Contents/Resources/Java/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr51/libc.a(isspace.o): In function `isspace':
(.text.avr-libc+0x2): relocation truncated to fit: R_AVR_13_PCREL against symbol `__ctype_isfalse' defined in .text.avr-libc section in /Applications/Arduino 1.6.app/Contents/Resources/Java/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr51/libc.a(cty_isfalse.o)
collect2: error: ld returned 1 exit status
Error compiling.

ryanneve, 2015-03-27:
This is really causing me a big problem. Is it an arduino problem a GCC problem? Ill that flash memory, and I can't use it.

ffissore, 2015-03-27:
If your project does an extensive use of PROGMEM and it's not worth porting it to IDE 1.6.x, then you should really stick with 1.0.x. It will remain available for download, imho, forever, it just won't receive updates

ryanneve, 2015-03-27:
I did get some confirmation that this is a gcc bug and there's a patch. Not to try and figure out how to match my gcc.
I believe my program compiles on 1.0.x not because the bug doesn't exist, but because of the way the linker has chosen to arrange things. It's just luck. I would hate to go back to the hundreds of `warning: only initialized variables can be placed into program memory area` messages I get on 1.0.x.

ryanneve, 2015-03-28:
Federico, Thanks for your guidance. I've followed the directions at:
https://github.com/arduino/toolchain-avr and installed cygwin and MinGW. They each seem to have their own environments and I'm at a loss on how to use MinGW from cygwin.

When I try `./arch.win32.build.bash` it fails and looking in config.log I see:
`mingw32-gcc command not found`. 
As far as I can tell, cygwin isn't aware of anything outside of it's install directory,(C:\cygwin) and MinGW installs  in C:\MinGW. Should I have installed MinGW in C:\cygwin?

I tried installing the MinGW  packages from within cygwin, but I end up with an executable called i686-w64-mingw32-gcc.exe

ryanneve, 2015-09-08:
This was bug 33698 in AVR C and was committed to avr-libc at revision 2473.

arjenhiemstra, 2015-12-04:
Any update on this issue?

ryanneve, 2016-03-07:
There is an [open bounty ](https://www.bountysource.com/issues/9447586-1-6-1-sketches-fail-to-link-with-r_avr_13_pcrel-due-to-interrupt-service-routine-too-far-for-startup-rjmp-in-vector)on this issue. First one that solves it opening a valid pull request wins 50$

cmaglie, 2016-07-01:
@ryanneve would you like to check if the build on #5021 solves the issue for you too?

ryanneve, 2016-07-01:
Cristian,

I'm not set up to build the IDE on Windows.

Sorry, I'd love to test it.

-Ryan

[image: Planktos Instruments]

Ryan Neve
 Ryan@PlanktosInstruments.com
 www.planktosinstruments.com
 (252) 515-0012

*CONFIDENTIALITY:*

This electronic transmission, including all attachments, is directed in
confidence solely to the person(s) to whom it is addressed, or an
authorized   recipient, and may not otherwise be distributed, copied or
disclosed.  The contents of the transmission may also be subject to
intellectual property rights and such rights are expressly claimed and are
not waived.  If you have received this transmission in error, please notify
the sender immediately by return electronic transmission and then
immediately delete this transmission, including all attachments, without
copying, distributing or disclosing same.

facchinm, 2016-07-01:
Hi Ryan,
you simply need to grab the compiled IDE from here https://github.com/arduino/Arduino/pull/5021#issuecomment-227791213 and you are ready to test :smile:

ryanneve, 2016-07-01:
> you simply need to grab the compiled IDE from here #5021 (comment)

Martino,

Not only does it fix my problem, but my program is much smaller. It went
from 44% to 37% of my mega2560's memory. Thanks, I can finally stop using
1.0.6,

-Ryan

[image: Planktos Instruments]

Ryan Neve
 Ryan@PlanktosInstruments.com
 www.planktosinstruments.com
 (252) 515-0012

*CONFIDENTIALITY:*

This electronic transmission, including all attachments, is directed in
confidence solely to the person(s) to whom it is addressed, or an
authorized   recipient, and may not otherwise be distributed, copied or
disclosed.  The contents of the transmission may also be subject to
intellectual property rights and such rights are expressly claimed and are
not waived.  If you have received this transmission in error, please notify
the sender immediately by return electronic transmission and then
immediately delete this transmission, including all attachments, without
copying, distributing or disclosing same.

cmaglie, 2016-07-27:
Fixed by #5021

closed 2016-07-27 by cmaglie