arduino/Arduino #2753: Weird Graphical Issues

SolusSoftworks, 2015-03-11:
I get weird UI glitches when trying to use the arduino software. Parts of the window are repeated and shifted making the software unusable. Here is what it looks like: 
[image]
I'm using the Windows 10 technical preview build 9926 but haven't had an issue with any other program.

cmaglie, 2015-03-11:
Are you running the latest Arduino IDE 1.6.1? Have you downloaded the .ZIP or the installer?
Is this happening with other old versions too?

SolusSoftworks, 2015-03-11:
Yes 1.6.1 but also occurred in previous versions. I used the installer. I've just downloaded the ZIP version which seems to be working so far.

cmaglie, 2015-03-11:
Thanks for the report.

Have you tried also with 1.0.5 and 1.0.6? If not, may I ask you to try with that versions too?

SolusSoftworks, 2015-03-11:
It happens with both of those too.

ffissore, 2015-03-20:
Can you give a try to the latest nightly build? This bug should be fixed by 8da868d47adf7c4e3aaffdf61b94287745e77ad6 http://arduino.cc/en/Main/Software#nightly

closed 2015-05-12 by ffissore