arduino/Arduino #2754: Virus in Arduino 1.6.1 ? (TR/Crypt.XPACK.Gen, found by Avira)

VogonJeltz, 2015-03-11:
Virus in Arduino 1.6.1 ?

Beginne mit der Suche in 'C:\Programme\Arduino\hardware\tools\avr\bin\avr-run.exe'
C:\Programme\Arduino\hardware\tools\avr\bin\avr-run.exe
[FUND] Ist das Trojanische Pferd TR/Crypt.XPACK.Gen
[HINWEIS] Eine Sicherungskopie wurde unter dem Namen 568eeae5.qua erstellt ( QUARANTÄNE )
[HINWEIS] Die Datei wurde ins Quarantäneverzeichnis unter dem Namen '4e19c5bd.qua' verschoben!

cmaglie, 2015-03-11:
Thanks for the report. The file has been already submitted to Avira that confirmed that it's a false positive, the IDE is safe to install:

https://analysis.avira.com/en/status?uniqueid=FBfEwP54zBzM32GUg2pFEZASV36qrvPh&incidentid=1839342

closed 2015-03-11 by cmaglie

VogonJeltz, 2015-03-11:
ok, thanks!
Now as Avira has already erased  the file avr-run.exe  from my HD - 
what is this file avr-run.exe for?
Will Arduino Sketch run without avr-run.exe ?

cmaglie, 2015-03-11:
As far as I know `avr-run` is an AVR simulator and it's not used by the IDE, so it should work even without it.

VogonJeltz, 2015-03-11:
thanks! (y)