arduino/Arduino #2755: Source modifications lead to compile failures

supuflounder, 2015-03-11:
I have several times gotten compilation errors compiling perfectly valid C++ code.  I am told that this is because the IDE, to simplify life for newbies, does all kinds of stuff to my source before sending it to the compiler.

I want an "expert mode" that says "hands off my code" and avoids the transformations that create erroneous programs out of correct programs

ryanneve, 2015-03-12:
You may want to look into using Atmel Studio instead of the Arduino IDE. Look into Visual Micro as well. A huge step up from the basic IDE. Unfortunately, I believe these are both windows only.

ffissore, 2015-03-12:
@supuflounder you can avoid the IDE magic by adding .h and .cpp "tabs" (files) to your sketch. They won't be preprocessed
If you want to help us improve the IDE preprocessor, please give a spin to the IDE linked at the bottom of this page https://github.com/arduino/Arduino/pull/2729. It solves all the preprocessor related issue reported so far

supuflounder, 2015-03-12:
Thank you, I will try it within the next few days.
        Joe

supuflounder, 2015-03-22:
Well, it doesn't work.  I get a null pointer exception in the preprocessor module (see below)
[code]
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/coan source -m -E -P -kb -DF_CPU=16000000L -DARDUINO=10602 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR /var/folders/qk/48tgmmgj3bq6x7_r2cxw_20w0000gn/T/sketch9131219889757286104.cpp
Using library Adafruit_NeoPixel in folder: /Users/flounder/Documents/Arduino/libraries/Adafruit_NeoPixel (legacy)
Using library colorref4 in folder: /Users/flounder/Documents/Arduino/libraries/colorref4 (legacy)
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzn /var/folders/qk/48tgmmgj3bq6x7_r2cxw_20w0000gn/T/sketch5858252572529864852.cpp
processing.app.debug.RunnerException
	at processing.app.debug.Compiler.compile(Compiler.java:359)
	at processing.app.debug.Compiler.build(Compiler.java:98)
	at processing.app.Sketch.build(Sketch.java:1169)
	at processing.app.Sketch.build(Sketch.java:1142)
	at processing.app.Editor$BuildHandler.run(Editor.java:1979)
	at java.lang.Thread.run(Thread.java:695)
Caused by: java.lang.NullPointerException
	at processing.app.preproc.CTagsParser.addPrototypes(CTagsParser.java:94)
	at processing.app.preproc.CTagsParser.preprocess(CTagsParser.java:38)
	at processing.app.preproc.CTagsBakedPreprocessor.parseCTagsOutput(CTagsBakedPreprocessor.java:113)
	at processing.app.preproc.CTagsBakedPreprocessor.preprocess(CTagsBakedPreprocessor.java:52)
	at processing.app.preproc.CTagsBakedPreprocessor.preprocess(CTagsBakedPreprocessor.java:42)
	at processing.app.debug.Compiler.compile(Compiler.java:357)
	... 5 more
[/code]

If you already know about this, then I'll wait for the fix; if not, I can send you the source code that causes the problem (contact me as newcomer@flounder.com).

ffissore, 2015-09-11:
Fixed by #3779

closed 2015-09-11 by ffissore