arduino/Arduino #2758: Error in LiquidCrystal - setCursor() example

YoniAbraham, 2015-03-12:
In the above example there is an error in the main program (loop). Below you can find the details (search for Error word in the comments):
void loop() {
  // loop from ASCII 'a' to ASCII 'z':
  for (int thisLetter = 'a'; thisLetter <= 'z'; thisLetter++) {
    // loop over the columns:
    for (int thisCol = 0; thisCol < numRows; thisCol++) {
      // loop over the rows:
      for (int thisRow = 0; thisRow < numCols; thisRow++) {
        // set the cursor position:
        //lcd.setCursor(thisCol, thisRow); // Error
        lcd.setCursor(thisRow, thisCol); // OK
        // print the letter:
        lcd.write(thisLetter);
        delay(200);
      }
    }
  }
}

ffissore, 2015-03-12:
The correct signature is setCursor(col, row). See #2114

closed 2015-03-12 by ffissore

cmaglie, 2015-03-12:
Probably what we really want is to rename `thisCol` to `thisRow` and vice-versa