arduino/Arduino #2759: Three connections missing in all of LiquidCrystal examples' electrical drawings

YoniAbraham, 2015-03-12:
the connections that are missing are:
1) "LED-" of the LCD should be connected to ground (GND).
2) "LED+" of the LCD should be connected to pin 1 of 220 Ohm resistor.
3) Pin 2 of the 220 Ohm resistor should be connected to 5V.
Note: the picture of breadbpard connections includes these connections.

agdl, 2015-03-12:
@YoniAbraham can you please post some link?
In all these the connections are right
http://arduino.cc/en/Tutorial/LiquidCrystalBlink
http://arduino.cc/en/Tutorial/LiquidCrystalCursor
http://arduino.cc/en/Tutorial/LiquidCrystalDisplay
http://arduino.cc/en/Tutorial/LiquidCrystalTextDirection
http://arduino.cc/en/Tutorial/LiquidCrystalScroll
http://arduino.cc/en/Tutorial/LiquidCrystalSerial
http://arduino.cc/en/Tutorial/LiquidCrystalSetCursor

agdl, 2015-03-27:
reopen if needed

closed 2015-03-27 by agdl