arduino/Arduino #2761: GUI glitch

Nerijuslt4, 2015-03-12:
My arduino GUI is glitching. It works, but it is really anoying when text is disappearing. I dont know how to explain. So the pictures should give more info: 
In first pic we see almost normal GUI, but these white dots doesnt seem right...
[image]
Then the fun begins, I sweep my mouse over text (i mean "file, edit, sketch...") and its gone. Well, almost gone... Few symbols stays sometimes.
[image]
Then even i'm getting even more fun: i press "File" in the corner and move my mouse around options, and they are disappearing too:
[image]

So please, help me. I can't work like this... Everything else works fine i gues. I  mean code writing, compiling, uploading etc. But when I need to open something from sketchbook, I'm almost guesing what I'm gonna open, cause file names are becoming invisible whan I move my mouse over them (and they are not comming back until i press something else).
It goes the same way with all other options (edit, sketch, etc.).

All started after i updated my arduino to 1.6.1 yesterday

What I have tried: reinstalling arduino, downloading zip file, updating java (idk if that has anything to do with arduino...), restarting computer. 
Any suggestions?

jcarlosgarciauah, 2015-03-14:
I have got the same problem, but in my case I have noticed the following:
* The IDE versión 1.6.1 Works fine within 'Windows 8.1'
* But the IDE-1.6.1 fails (in the way you show here) within 'Windows 7'

Is this the case with you? Are you running Windows-7?

Nevertheless, previous versions of IDE Works fine with Windows 7. I think this is a bug in IDE-1.6.1, let us hope that versión 1.6.2 (coming son) will address this bug!

Nerijuslt4, 2015-03-14:
Yes, I'm using win-7. And yes, previous versions worked fine.

ffissore, 2015-03-17:
Can you please give a try to this build and tell if it fixes your issue? https://github.com/arduino/Arduino/pull/2783#issuecomment-82223032

kloper, 2015-03-17:
Same issue here. Win7 64 bit.

Nerijuslt4, 2015-03-18:
Had only few free minutes and tried
http://arduino.cc/download.php?f=/javaide/pull_requests/arduino-PR-2783-BUILD-224-windows.zip
Those issues seems like to be gone at first look. Ill test it more later.

commit 8da868d4 (2015-03-19): Attempt to fix glitches introduced by java8 on windows. Fixes #2761
commit dcb65d9e (2015-03-19): Merge pull request #2783 from ffissore/windows-glitch

Attempt to fix glitches introduced by java8 on windows. Fixes #2761
closed 2015-03-19 by ffissore