arduino/Arduino #2762: Compilation error: missing terminating " character

qbahn21, 2015-03-12:
Code sample:
String s =
 "\\"
 "//";
Serial.println ( s );

Regards,
Qbahn21

qbahn21, 2015-03-12:
applies to Arduino 1.6.1

ffissore, 2015-03-13:
That's because of the preprocessor and there already is a fix for that: please give a spin to this version of the IDE #2729 (bottom of the page)

qbahn21, 2015-03-13:
Thanks, but I don't know what a spin is... :(

ffissore, 2015-03-13:
Sorry, I meant: can you please download a build of the IDE suitable for your operating system from #2729 (bottom of the page) and try using it to compile the sketch you posted? That build should contain a fix to your issue

qbahn21, 2015-03-13:
Done already. That build works fine with that sample. But whole sketch doesn't compile. Hangs up during "coan source -m -E -P -kb -DF_CPU=16000000L -DARDUINO=10602 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR..." without any additional output :(

closed 2015-03-13 by qbahn21

ffissore, 2015-03-13:
Can you share the whole sketch?