arduino/Arduino #2765: How do you know which tab you're editing?

clefranc, 2015-03-13:
Hi all,
Sorry if this question was asked before, but I can't find an answer in the thousand of closed and opened issues.

Anyway, I apologize in advance for my ignorance.

I'm working on a "sketch" with 100+ files, each with its own tab. The problem is to determine which tab I'm currently editing. The tabs are fixed, they don't scroll, there is no indication in the title bar or the status bar, no indication in the scrolling menu (arrow), etc. Even with my 4096 x 2160 monitor I can't see all the tabs.

So, how do you know which tab you're editing.

I apologize again for my blatant ignorance.

Christian

ffissore, 2015-03-13:
Thanks for the report Christian, I've never made such a huge sketch. This
missing highlighting is something we must fix

revwarguy, 2015-03-26:
+1 on this - I have about 40 files (tabs) and having the current filename somewhere on the screen would be helpful

clefranc, 2015-04-24:
Any news on this?

clefranc, 2015-06-08:
Apparently not! I envy you to only use a sketch with 2 tabs...

ffissore, 2015-06-25:
Fixed. Fix will be available in next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-06-25 by ffissore

clefranc, 2015-06-25:
@ffissore Thank you very much!

lmihalkovic, 2015-11-21:
@clefranc  I could use UX feedback from your 'extreme sketch'. My current solution for the problem has been to 
* allow sub-folders with code inside a sketch
* add a outline view to the IDE

So far it has answered my own needs, but perhaps it would still be not even close to making your life easier?! Let me know if you have any "IF ONLYs"... no promise, but I will look into it.

You can find out more in #4083

commit 2339049c (2015-06-25): When switching between tabs, add the name of the current file/tab to the title. Fixes #2765