arduino/Arduino #2770: Better error message if sketch too big

NicoHood, 2015-03-14:
Used Arduino Leonardo Board and IDE 1.6.1 on Windows 8.1

```
Sketch uses 31,416 bytes (109%) of program storage space. Maximum is 28,672 bytes.
Global variables use 1,335 bytes (52%) of dynamic memory, leaving 1,225 bytes for local variables. Maximum is 2,560 bytes.
processing.app.debug.RunnerException: Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
	at processing.app.debug.Compiler.size(Compiler.java:325)
	at processing.app.debug.Compiler.build(Compiler.java:117)
	at processing.app.Sketch.build(Sketch.java:1169)
	at processing.app.Sketch.build(Sketch.java:1142)
	at processing.app.Editor$BuildHandler.run(Editor.java:1977)
	at java.lang.Thread.run(Thread.java:745)
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
```

matthijskooijman, 2015-03-17:
What do you think needs improving? The error message looks pretty ok to me, except that the backtrace doesn't really serve any purpose.

NicoHood, 2015-03-17:
The output. Is it intended that it posts that errors?
```
processing.app.debug.RunnerException: Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
    at processing.app.debug.Compiler.size(Compiler.java:325)
    at processing.app.debug.Compiler.build(Compiler.java:117)
    at processing.app.Sketch.build(Sketch.java:1169)
    at processing.app.Sketch.build(Sketch.java:1142)
    at processing.app.Editor$BuildHandler.run(Editor.java:1977)
    at java.lang.Thread.run(Thread.java:745)
```