arduino/Arduino #2772: Compile error when declaring a pointer-function

happybrick, 2015-03-14:
I get a weird compile-error with self-written classes when i declare a function that returns a pointer to that very class. Can't figure out why.
UECIDE 0.8.7z36 compiles the example-code (ardubug.ino) without any errors though.
Using: Windows7-64bit, Arduino-IDE 1.6.1, Arduino Nano, ATmega328 ( Same problem for all other platines / prozessors, too )

--- Arduino-IDE - Error-Output:
ardubug.ino:2:1: error: 'MyClass' does not name a type
Fehler beim Kompilieren.

--- Code:
class MyClass{
  public:
    MyClass();
    int value;
};

MyClass::MyClass(){
  this->value = 101;
}

MyClass* someProcedure(){
  return NULL;
}
void setup() {}
void loop() {}

ffissore, 2015-03-16:
Hi
this is an issue with the current preprocessor. There is an experimental build that fixes your issue: you may want to give it a try. See #2729 (bottom of the page). This build is not perfect, but works with most simple sketches

happybrick, 2015-03-16:
Thx, i will definitly will give it a try.

closed 2015-03-16 by happybrick