arduino/Arduino #2773: 1.6.1 Ate my sketches

knildon, 2015-03-14:
I updated my software to 1.6.1 and my sketches were deleted even though it said they would not be deleted.  All of the sketches I have generated were saved in whichever file the Arduino software puts them.  I have never created a special file to save my sketches and didn't know I had to if I did.
Is this a glitch in the software or did I do something wrong?
I was able to get some of the sketches back from the trash bin but some were completely gone!!
I would appreciate some feedback on this so I can prevent future losses or maybe the software can be revised to prevent this from happening again.
Don

PaulStoffregen, 2015-03-14:
Normally, deleting the IDE should not erase your sketches.  They're stored in a separate location called the "sketchbook".

However, perhaps you had the sketchbook location incorrectly configured to be the same directory as your Arduino installation?  This is already filed as issue #2719 

Could you look in File > Preferences and tell us what directory is configured for your sketchbook?  And of course, what directory is actually being used for Arduino?  I realize on Windows and modern Macs, many of the graphical programs for managing files try to "simplify" file systems by obscuring locations, so it may not be easy for you to get accurate info about where things really are on your computer's filesystem.  If File > Preferences, you can turn on verbose output while compiling, which will print compiler commands when you Verify or Upload.  Some of those will have the real, full pathnames.

If we're ever going to prevent this problem from wiping out anyone else's files in the future, we've really got to collect accurate info about exactly where you had things installed.  My hunch is this may be the already-known case of issue #2719, but really that's just a blind guess.

I've seen a couple other comments on the Arduino forum about something similar happening.  Maybe that was also you?  Or maybe this is a problem that's hitting multiple people?  I really want to figure out what happened and eventually add code to Arduino to make sure nobody else suffers this fate.  But until we understand why, the best I can say is maybe it's #2719, or maybe it's some other misconfiguration nobody has understood yet?  I know I've deleted and reinstalled Arduino many hundreds of times while developing features and testing, and I've never managed to stumble across this situation... so I really need details from you.

knildon, 2015-03-15:
I checked the hard drive and found an Arduino folder in "My Documents" and it had all the files except a "sketchbook" folder.  I also found an Arduino folder in "Programs" and it only had an uninstall file in it.
I decided to install 1.6.0 on my wife's laptop to see what would happen.  The install placed an Arduino folder in"Programs" with all the files except a "sketchbook" folder.  I also found an Arduino folder in "My Documents" with a "libraries" folder in it.  The "libraries" folder had a readme file in it.
When I tried to run the program I received a message that the sketchbook was not there.  I don't remember the whole message since it only did it the one time.  I generated a test sketch and saved it.  It was put in the Arduino folder in "My Documents".  I still don't have a sketchbook folder.
I also installed 1.6.1 and it was installed exactly the same as 1.6.0 was and my test sketch was not deleted.  Still no sketchbook folder.  
The only thing that I can guess, since I don't remember, is that when I installed the original program I told the program where to be installed.  Even if that were the case it still should have separated the files like it did on a new install.  This much I do remember, I didn't tell it where to be installed when I upgraded to 1.6.1.
I hope this info helps so on your next upgrade you can inject an "Idiot proof" routine, for people like me, in your program.
Don

PaulStoffregen, 2015-03-15:
Sounds like everything is works as it should.

If you have access to another clean, previously unused machine (or a virtual machine), you'd need to retrace (and document) your original steps.  That's what would be needed to figure out what went wrong and what could be done in future versions to prevent it.

knildon, 2015-03-15:
I completely removed all Arduino programs, after I put all of my sketches and libraries in a save file and then reloaded 1.6.1.  I directed the program to be loaded in the "Programs" directory.  Once that was done, I copied my files back into the proper directories and all is working well.  The one part that confused me is that I was looking for a folder named "Sketchbook" in the Arduino folder.  It appears that only exists in the program menu.  The sketches are actually saved in the Arduino folder in the "My Documents" directory.
I'm still puzzled as to why I can't tell the program where to reside on my machine.  No matter where it resides it still should generate 2 Arduino files and place all of the files in their respective directories/folders. It didn't do that on my machine.  it put everything in one directory as I stated before.
Thanks for your quick reply and let me know if you come up with any other conclusions to my dilemma.
Don

closed 2015-05-29 by ffissore