arduino/Arduino #2774: No space on the left side of the code

chq76, 2015-03-15:
Hello,

since years there is a small problem with arduino on mac. Sometimes, after wrting a lot of code, there is suddenly no space between the code and the left border.

Here you can see a small movie, showing this problem:

https://www.sendspace.com/file/r0enjb

Months ago i wrote this issue in the arduino forum, but nothing happened.

The bug is still present (in 1.6.1 too).

Greetz Chris

ffissore, 2015-03-16:
Does enabling line numbers works around the issue?

chq76, 2015-03-16:
When i close arduino and open it again, everything is fine again. I don't want to use line numbers.

Greetz Chris

ffissore, 2015-03-17:
I wasn't suggesting to keep line numbers on. I was asking if, with line numbers enabled, the issue disappears or is still there. This could help us understanding the root cause of the issue

chq76, 2015-03-17:
Don't know, because i don't use enabled line numbers.

Greetz Chris

ffissore, 2015-03-20:
Chris, please strictly follow these steps: 1) enable line numbers, 2) verify if the space issue is still present, 3) disable line numbers

chq76, 2015-03-20:
Please consult your betatesters.

Greetz Chris

closed 2015-04-09 by ffissore

chq76, 2015-04-09:
Why closed? Is there a solution?

ffissore, 2015-04-09:
No, but I wasn't able to reproduce the issue, nor to convince you trying if enabling line number worked around it