arduino/Arduino #2775: ATtiny in Arduino 1.6.1

wdreumel, 2015-03-15:
Since I switched to 1.6.1, ATtiny does not show up in the board list.
I just published a book based on a previous version. So almost one thousand readers will experience the same problem. I placed the map attiny in the hardware map without result.
Please help.

Lauszus, 2015-03-15:
@wdreumel you should take a look at: https://github.com/arduino/Arduino/wiki/Arduino-Hardware-Cores-migration-guide-from-1.0-to-1.6 and https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification.

ffissore, 2015-03-16:
@wdreumel take a look at https://github.com/damellis/attiny/ There is a specific zip file for IDE 1.6 and attiny https://github.com/damellis/attiny/archive/ide-1.6.x.zip

sej7278, 2015-03-28:
i found that the damellis 1.6 branch doesn't compile at all, the arduino-tiny 1.5 branch does

damellis, 2015-03-28:
What error are you getting with https://github.com/damellis/attiny/tree/ide-1.6.x? I just tried it and it seems to work fine (on Mac, at least).

sej7278, 2015-03-28:
```
~/arduino-1.6.1/hardware/tools/avr/avr/include/util/delay.h: In function 'void _delay_ms(double)':
~/arduino-1.6.1/hardware/tools/avr/avr/include/util/delay.h:150:17: error: expected primary-expression before ')' token
  __tmp = ((F_CPU) / 1e3) * __ms;
                 ^
~/arduino-1.6.1/hardware/tools/avr/avr/include/util/delay.h: In function 'void _delay_us(double)':
~/arduino-1.6.1/hardware/tools/avr/avr/include/util/delay.h:232:17: error: expected primary-expression before ')' token
  __tmp = ((F_CPU) / 1e6) * __us;
                 ^
```

damellis, 2015-03-28:
Hmm...  which board, processor, and clock do you have selected from the Tools menu? This is with the 1-6.x tree, right?

damellis, 2015-03-28:
Do you have anything selected in the Tools > Clock menu? That might be the problem.

sej7278, 2015-03-28:
yes this is 1.6.x tree, it seems to work ok in the 1.6.1 ide

i think my problem is my makefile, its not finding the tiny8 variant directory when using:

```
BOARD_TAG = attiny
BOARD_SUB = attiny85
```

cmaglie, 2015-04-07:
It seems that this issue has a solution, so I'm closing it. Feel free to reopen if needed.

closed 2015-04-07 by cmaglie